apache/nuttx-apps #246: Illegal calls to romdisk_register()

patacongo, 2020-05-10:
Several examples (and other things under apps/) make illegal calls to `romdisk_register()`.  This both violates the portable POSIX OS interface and makes these applications un-usable in PROTECTED and KERNEL build modes.

Non-compliant examples include:

    examples/bastest, examples/elf, examples/module,
    examples/nxflat, examples/posix_spawn, examples/romfs,
    examples/sotest, examples/thttpd, examples/unionfs

These examples are simple demos and, hence, you could argue that it is not so bad that they violate the interface for the purpose of demonstration (although they do set a bad example because of this).

These examples should, of course, use boardctl(BOARDIOC_ROMDISK)  to create the ROM disk instead of calling `romdisk_register() `directly.

patacongo, 2020-05-10:
Refer to Issue #245 for a similar improper OS call that needs to be fixed.

yamt, 2020-08-24:
this look like a dup of https://github.com/apache/incubator-nuttx-apps/issues/139

tanushreebaindur, 2021-03-27:
I am new to Nuttx and wanted to know if I can take up this issue. Can the code changes (that is replacing romdisk_register() with boardctl(BOARDIOC_ROMDISK) in all of the non-compliant examples) be tested using the simulator (nsh:sim) configuration or do you recommend using a board for testing?

xiaoxiang781216, 2021-03-27:
Yes, thanks for heping @tanushreebaindur. Ether sim or real board is fine since it's a common feature not depend on the particular hardware.

tanushreebaindur, 2021-03-28:
I got the master branch version of nuttx and nuttx apps a few days ago and I am using the nsh:sim configuration. 

Using ‘make menuconfig’ I enabled the Memory Manager(MM) debug features including the error, warning and informational output. I then typed ‘make’ and then ran the simulator. This caused the simulator to exit without going to the nsh prompt due to an assertion failing as described below:

Following is part of the output I see  :
mm_free: Freeing 0x10bd5aef0
mm_free: Freeing 0x10bd5af10
mm_free: Freeing 0x10bcdad90
mm_free: Freeing 0x10bcdadd0
mm_free: Freeing 0x10bcdad70
login: mm_free: Freeing 0x10bcd8ef0
Assertion failed at file:mm_heap/mm_checkcorruption.c line: 101

Line 101 in mm_checkcorruption.c is:
             assert(node->size >= SIZEOF_MM_FREENODE);

Running the simulator without the mm debug features enabled will not cause the above problem. The simulator works fine and comes to the nsh prompt and I am able to run the hello world app and also the examples/bastest app at the nsh prompt.

I wanted to know if I am missing something as the assertion is failing on enabling the mm debug features and not otherwise.

closed 2021-05-12 by xiaoxiang781216

tanushreebaindur, 2021-05-12:
@xiaoxiang781216 examples/nxflat and examples/thttpd still contain the illegal calls to the romdisk_register(). I wasn't able to test those on the simulator or board so they are still unchanged. Please let me know if I need to open a new issue to track those 2 examples.

This is the message thread for reference: http://mail-archives.apache.org/mod_mbox/nuttx-dev/202104.mbox/%3CCABHE6qt1K5P_SNMhZNB6tKba-nqaK1EmfOseVfqRfymdQZwhkQ%40mail.gmail.com%3E

xiaoxiang781216, 2021-05-12:
Since the modification is simple, the change pass the CI  should be OK to merge.

tanushreebaindur, 2021-05-13:
> Since the modification is simple, the change pass the CI should be OK to merge.

Ok. I will make the changes to examples/nxflat and examples/thttpd and submit another PR.

a-lunev, 2021-05-19:
Hello,

commit ab41d23d7c1d89ae59aa550aa0755adb5c9e7bea broke eagle100:nxflat config:
```
CC:  nxflat_main.c
.../buildroot-gcc-7.4.0/build_arm_nofpu/staging_dir/bin/arm-nuttx-eabi-gcc -c -fno-builtin -Wall -Wstrict-prototypes -Wshadow -Wundef -Os -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -isystem ".../nuttx/include" -D__NuttX__  -pipe -I ".../apps/include" -Dmain=nxflat_main  nxflat_main.c -o  nxflat_main... ...apps.examples.nxflat.o
nxflat_main.c: In function 'nxflat_main':
nxflat_main.c:157:29: error: storage size of 'desc' isn't known
   struct boardioc_romdisk_s desc;
                             ^~~~
nxflat_main.c:168:9: warning: implicit declaration of function 'boardctl'; did you mean 'prctl'? [-Wimplicit-function-declaration]
   ret = boardctl(BOARDIOC_ROMDISK, (uintptr_t)&desc);
         ^~~~~~~~
         prctl
nxflat_main.c:168:18: error: 'BOARDIOC_ROMDISK' undeclared (first use in this function); did you mean '_BOARDIOCVALID'?
   ret = boardctl(BOARDIOC_ROMDISK, (uintptr_t)&desc);
                  ^~~~~~~~~~~~~~~~
                  _BOARDIOCVALID
nxflat_main.c:168:18: note: each undeclared identifier is reported only once for each function it appears in
nxflat_main.c:157:29: warning: unused variable 'desc' [-Wunused-variable]
   struct boardioc_romdisk_s desc;
                             ^~~~
```

The last commit w/o the build error was b7e9c434891611f0f0d0b2ea683bf4043e629107.

Steps to reproduce:
```
$ mkdir TEST_ROOT
$ git clone https://github.com/apache/incubator-nuttx.git TEST_ROOT/nuttx
$ git clone https://github.com/apache/incubator-nuttx-apps TEST_ROOT/apps
$ cd TEST_ROOT/nuttx
$ ./tools/configure.sh -l eagle100:nxflat
```

Build NXFLAT Toolchain:
```
$ git clone https://bitbucket.org/nuttx/buildroot.git TEST_ROOT/buildroot
$ cd TEST_ROOT/buildroot
$ cp configs/cortexm3-eabi-defconfig-7.4.0 .config
$ make oldconfig
$ make
```

Build NuttX:
```
$ cd TEST_ROOT/nuttx
$ make CROSSDEV=TEST_ROOT/buildroot/build_arm_nofpu/staging_dir/bin/arm-nuttx-eabi- \
       MKNXFLAT=TEST_ROOT/buildroot/build_arm_nofpu/staging_dir/bin/mknxflat \
       LDNXFLAT=TEST_ROOT/buildroot/build_arm_nofpu/staging_dir/bin/ldnxflat
```

tanushreebaindur, 2021-05-19:
Sorry about that. I see that CONFIG_BOARDCTL_ROMDISK is not enabled /nuttx/boards/arm/tiva/eagle100/configs/nxflat/defconfig

CONFIG_BOARDCTL_ROMDISK can be enabled by doing the following in menuconfig: BOARD SELECTION -> Enable application space creation of ROM disks.

I will submit a PR soon with the updated nuttx/boards/arm/tiva/eagle100/configs/nxflat/defconfig file.

a-lunev, 2021-05-19:
Hi @tanushreebaindur,

Possibly an additional check would be useful in nxflat_main.c along with the other existing checks:
```
#ifndef CONFIG_BOARDCTL_ROMDISK
#  error "CONFIG_BOARDCTL_ROMDISK should be enabled in the configuration file"
#endif
```

tanushreebaindur, 2021-05-19:
I accidentally deleted this comment:

Sorry about that. I see that CONFIG_BOARDCTL_ROMDISK is not enabled /nuttx/boards/arm/tiva/eagle100/configs/nxflat/defconfig

CONFIG_BOARDCTL_ROMDISK can be enabled by doing the following in menuconfig: BOARD SELECTION -> Enable application space creation of ROM disks.

I will submit a PR soon with the updated nuttx/boards/arm/tiva/eagle100/configs/nxflat/defconfig file.

tanushreebaindur, 2021-05-19:
> Hi @tanushreebaindur,
> 
> Possibly an additional check would be useful in nxflat_main.c along with the other existing checks:
> 
> ```
> #ifndef CONFIG_BOARDCTL_ROMDISK
> #  error "CONFIG_BOARDCTL_ROMDISK should be enabled in the configuration file"
> #endif
> ```

Sure, I will add that check to nxflat_main.c. Thanks.

a-lunev, 2021-05-21:
Hi @tanushreebaindur,
I've tested commit 2217d2f0d34c67e30ac32f089410308a6ca6a061. Now the build fails with the following error:
```
../nuttx/arch/arm/src/board/libboard.a(lm_appinit.o): In function `board_app_initialize':
lm_appinit.c:(.text+0x46): undefined reference to `mmcsd_spislotinitialize'
make[1]: *** [Makefile:156: nuttx] Error 1
```
Steps to reproduce are the same as I previously wrote.

tanushreebaindur, 2021-05-21:
Hi @a-lunev ,

The error is because CONFIG_MMCSD and CONFIG_MMCSD_SPI are not defined. mmcsd_spislotinitialize() is enabled when CONFIG_MMCSD and CONFIG_MMCSD_SPI are defined

I do not have the board and I'm currently using the simulator to make the configuration change.

Can you please make the following change to the configuration and let me know if it works?
I can then submit a PR for the corrected eagle100:nxflat defconfig

To enable CONFIG_MMCSD: make menuconfig --> Device Drivers --> MMC/SD Driver Support
To enable CONFIG_MMCSD_SPI: make menuconfig --> Device Drivers --> SPI Driver Support

This is the resulting defconfig after enabling CONFIG_MMCSD and CONFIG_MMCSD_SPI:

#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
CONFIG_ARCH="arm"
CONFIG_ARCH_BOARD="eagle100"
CONFIG_ARCH_BOARD_EAGLE100=y
CONFIG_ARCH_CHIP="tiva"
CONFIG_ARCH_CHIP_LM3S6918=y
CONFIG_ARCH_CHIP_LM3S=y
CONFIG_ARCH_CHIP_LM=y
CONFIG_ARCH_STACKDUMP=y
CONFIG_BOARDCTL_ROMDISK=y
CONFIG_BOARD_LOOPSPERMSEC=4531
CONFIG_CONSOLE_SYSLOG=y
CONFIG_DISABLE_ENVIRON=y
CONFIG_EXAMPLES_NXFLAT=y
CONFIG_FS_ROMFS=y
CONFIG_LIB_BOARDCTL=y
CONFIG_MAX_TASKS=16
CONFIG_MMCSD=y
CONFIG_NXFLAT=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAM_SIZE=65536
CONFIG_RAM_START=0x20000000
CONFIG_RAW_BINARY=y
CONFIG_RR_INTERVAL=200
CONFIG_SDCLONE_DISABLE=y
CONFIG_SPI=y
CONFIG_START_DAY=6
CONFIG_START_MONTH=5
CONFIG_START_YEAR=2009
CONFIG_SYMTAB_ORDEREDBYNAME=y
CONFIG_TASK_NAME_SIZE=0
CONFIG_TIVA_GPIOA_IRQS=y
CONFIG_TIVA_GPIOB_IRQS=y
CONFIG_TIVA_GPIOC_IRQS=y
CONFIG_TIVA_GPIOD_IRQS=y
CONFIG_TIVA_GPIOE_IRQS=y
CONFIG_TIVA_GPIOF_IRQS=y
CONFIG_TIVA_GPIOG_IRQS=y
CONFIG_TIVA_SSI0=y
CONFIG_TIVA_UART0=y
CONFIG_UART0_SERIAL_CONSOLE=y
CONFIG_USERMAIN_STACKSIZE=4096
CONFIG_USER_ENTRYPOINT="nxflat_main"

a-lunev, 2021-05-21:
Hi @tanushreebaindur,

There should be a choice to enable or not enable CONFIG_MMCSD and CONFIG_MMCSD_SPI options. The existing defconfig does not have the options enabled.
I have not investigated the issue by myself, however I suppose that error should be fixed by isolating the referencing code (according to the error it references mmcsd_spislotinitialize() from lm_appinit.c) by conditional compilation (#ifdef ... #endif) or some other way.

Concerning a physical board, I also do not have it. I can check only for build errors as I wrote in "Steps to reproduce" instructions.

tanushreebaindur, 2021-05-21:
Hi @a-lunev ,

With refernce to  your message:
> There should be a choice to enable or not enable CONFIG_MMCSD and CONFIG_MMCSD_SPI options. The existing defconfig does not have the options enabled.

Thanks for pointing this out. I did not think about it earlier.

I followed the instructions to reproduce the error. After this last step:
$ cd nuttx-code/nuttx
$ make CROSSDEV=nuttx-code/buildroot/build_arm_nofpu/staging_dir/bin/arm-nuttx-eabi- \
       MKNXFLAT=nuttx-code/buildroot/build_arm_nofpu/staging_dir/bin/mknxflat \
       LDNXFLAT=nuttx-code/buildroot/build_arm_nofpu/staging_dir/bin/ldnxflat

I get the following error on doing make:

make[3]: *** No rule to make target 'context'.  Stop.
make[3]: Leaving directory '/home/tanushree/nuttx-code/apps/examples/nxflat'
Makefile:61: recipe for target '/home/tanushree/nuttx-code/apps/examples/nxflat_context' failed
make[2]: *** [/home/tanushree/nuttx-code/apps/examples/nxflat_context] Error 2
make[2]: Leaving directory '/home/tanushree/nuttx-code/apps'
Makefile:163: recipe for target 'context' failed
make[1]: *** [context] Error 2
make[1]: Leaving directory '/home/tanushree/nuttx-code/apps'
tools/Makefile.unix:342: recipe for target 'context' failed
make: *** [context] Error 2

I have not made any change to the code and have got it from upstream. It looks like the rule to make context is missing. Did you encounter this error?

a-lunev, 2021-05-21:
@tanushreebaindur,
I tried the recent upstream state and I did not encounter that error.
I'm not sure about what namely VM you are using. What do you mean?

tanushreebaindur, 2021-05-23:
@a-lunev ,

The above error was my mistake. I had intentionally edited the nxflat/Makefile some days back while trying something and then deleted it  so that I could update it from upstream. But fetching and merging from upstream to my test branch did not get me a copy of the Makefile, so there was no Makefile in examples/nxflat. I corrected that problem by  cloning the repositories again. 

After that I got the following error on building Nuttx:
LD: nuttx
/home/tanushree/nuttx-code/buildroot/build_arm_nofpu/staging_dir/bin/arm-nuttx-eabi-ld: cannot find -lm
Makefile:155: recipe for target 'nuttx' failed
make[1]: *** [nuttx] Error 1
make[1]: Leaving directory '/home/tanushree/nuttx-code/nuttx/arch/arm/src'
tools/Makefile.unix:422: recipe for target 'nuttx' failed
make: *** [nuttx] Error 2

To fix the above error I had to enable CONFIG_LIBM using menuconfig (Library Routines --> Standard Math Library).  I used a 32 bit virtual machine for this build because I am getting multiple definition errors when building buildroot on the 64bit machine. But I did not have to make any other configuration changes due to using a 32 bit virtual machine instead of 64 bit. Adding CONFIG_LIBM was the only change.

I was then able to reproduce this error:

../nuttx/arch/arm/src/board/libboard.a(lm_appinit.o): In function `board_app_initialize':
lm_appinit.c:(.text+0x46): undefined reference to `mmcsd_spislotinitialize'
make[1]: *** [Makefile:156: nuttx] Error 1

To fix this error I isolated the call to mmcsd_spislotinitialize  by conditional compilation as you have suggested above. The code now builds successfully.

I  will submit a PR so the changes can be evaluated

a-lunev, 2021-05-23:
> Do you know if I can run the hello program under nxflat/tests/hello without a board? If not, I will submit a PR so the chnages can be evaluated.

I did not find how to emulate eagle100 board. To test nxflat examples I'm using QEMU that supports lm3s6965-e board for emulation. Please, check issue #3737 (https://github.com/apache/incubator-nuttx/issues/3737) for instructions and PR #3763 (https://github.com/apache/incubator-nuttx/pull/3763) that contains configuration file to test nxflat on lm3s6965-e board using QEMU.

tanushreebaindur, 2021-05-25:
Thank you. I was able to run nxflat examples using your instructions. 

To run the nxflat examples I used a 32 bit virtual machine (Virtual Box) to generate nuttx.bin and then used qemu-system-arm which I installed on the host computer.

Would you be able to point me to directions to install QEMU so I can configure and build nuttx for different configurations on it? That way I wouldn't need Virtual Box. Thanks again.

a-lunev, 2021-05-25:
Hi @tanushreebaindur,

I installed QEMU on my laptop (Debian 10) as follows:
`$ sudo apt-get install qemu-system-arm`

What host OS are you using?

tanushreebaindur, 2021-05-26:
Host OS is 64 bit Pop OS. It is based on Ubuntu. 

I used the same command (sudo apt-get install qemu-system-arm)to install QEMU and I was able to successfully run nxflat examples. I was wondering if you were also using QEMU as a virtualizer along with KVM to build nuttx.bin. This is because I am only able to build nxflat on a 32 bit machine (which is the 32 bit Ubuntu virtual machine I've installed on my 64 bit host OS.). When I tried to build nxflat on my 64 bit host OS, I get multiple definition errors.

a-lunev, 2021-05-26:
I did not use KVM for nuttx building. I built NXFLAT Toolchain using Debian 10 (x86_64). Then I built nuttx using NXFLAT Toolchain (based on gcc-7.4.0). All according to the instructions I provided before. I did not experience errors during the build.

a-lunev, 2021-05-26:
However, my Debian 10 is not freshly installed system. Previously I installed many other packages for other needs.

According to README (https://bitbucket.org/nuttx/buildroot.git) the following packages should be installed on the host OS:
```
sudo apt install build-essential
sudo apt install bison flex gperf
sudo apt install libgmp-dev libmpc-dev libmpfr-dev libisl-dev
sudo apt install binutils-dev libelf-dev
sudo apt install libexpat-dev
```

tanushreebaindur, 2021-05-29:
Thank you so much. I am able to build and run nuttx.bin on my 64 bit host machine. I checked and found that of the above packages are installed on my system. 

The problem was that I was getting the following build errors when building buildroot:

/usr/bin/ld: libsim.a(maverick.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/maverick.c:63: multiple definition of `DSPregs'; libsim.a(wrapper.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/wrapper.c:130: first defined here
/usr/bin/ld: libsim.a(maverick.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/maverick.c:65: multiple definition of `DSPsc'; libsim.a(wrapper.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/wrapper.c:132: first defined here
/usr/bin/ld: libsim.a(maverick.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/maverick.c:64: multiple definition of `DSPacc'; libsim.a(wrapper.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/wrapper.c:131: first defined here
/usr/bin/ld: libsim.a(armemu32.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/armemu.c:1145: multiple definition of `isize'; libsim.a(armemu26.o):/nuttx-code/buildroot/toolchain_build_arm_nofpu/gdb-8.0.1/sim/arm/armemu.c:1145: first defined here

To solve this problem, I made  code changes in the gdb-8.0.1/sim/arm described here: https://sourceware.org/legacy-ml/gdb-patches/2019-12/msg00250.html because I did not know how to download the patch. After that everything worked.