apache/nuttx-apps #340: Is preconfig necessary for application directories?

Ouss4, 2020-07-24:
This is somewhat related to some of the issues pointed out here: https://github.com/apache/incubator-nuttx/issues/1407

Most of the configuration phase output is coming from running the `preconfig` target inside all the application directories.
(By application directories I'm referring to the lowest level, for instance: apps/examples/adc)
That target is inherited from `Application.mk` and applications and examples could then extend it.
As of right now, none of the applications present in the repo use that target.  I also looked for some examples in other repositories but I couldn't find any.

The addition came as part of https://github.com/apache/incubator-nuttx-apps/commit/a9aa753fa4d282bc0aa48333ece593bdb5dc707b which added the dynamic top-level Kconfig generation.
This prepares the top-level Kconfig files, at the level of application directory there is no more preparation.  So I'm wondering if someone is using this for any other purpose?  If so, shouldn't that be part of `context`?

Removing this part will significantly reduce configuration time and also provides a cleaner output.
I have the following patch:

```diff
diff --git a/Application.mk b/Application.mk
index d23492eb..9549f7ac 100644
--- a/Application.mk
+++ b/Application.mk
@@ -96,7 +96,7 @@ VPATH += :.
 # Targets follow

 all:: .built
-.PHONY: clean preconfig depend distclean
+.PHONY: clean depend distclean
 .PRECIOUS: $(BIN)

 define ELFASSEMBLE
@@ -195,8 +195,6 @@ install::

 endif # BUILD_MODULE

-preconfig::
-
 ifeq ($(CONFIG_NSH_BUILTIN_APPS),y)
 ifneq ($(PROGNAME),)
 ifneq ($(PRIORITY),)
diff --git a/Directory.mk b/Directory.mk
index 66fbbdab..e04b414e 100644
--- a/Directory.mk
+++ b/Directory.mk
@@ -43,7 +43,6 @@ all: nothing

 .PHONY: nothing context depend clean distclean

-$(foreach SDIR, $(SUBDIRS), $(eval $(call SDIR_template,$(SDIR),preconfig)))
 $(foreach SDIR, $(SUBDIRS), $(eval $(call SDIR_template,$(SDIR),context)))
 $(foreach SDIR, $(SUBDIRS), $(eval $(call SDIR_template,$(SDIR),depend)))
 $(foreach SDIR, $(SUBDIRS), $(eval $(call SDIR_template,$(SDIR),clean)))
@@ -53,7 +52,7 @@ nothing:

 install:

-preconfig: $(foreach SDIR, $(SUBDIRS), $(SDIR)_preconfig)
+preconfig:
 ifneq ($(MENUDESC),)
        $(Q) $(MKKCONFIG) -m $(MENUDESC)
 endif
diff --git a/Make.defs b/Make.defs
index 92cae86a..e132c20e 100644
--- a/Make.defs
+++ b/Make.defs
@@ -50,11 +50,15 @@ endif
 # Application Directories

 # BUILDIRS is the list of top-level directories containing Make.defs files
+# CONFDIRS is the list of directories where that the preconfig target will
+#   be execute in.
 # CLEANDIRS is the list of all top-level directories containing Makefiles.
 #   It is used only for cleaning.

 BUILDIRS  := $(dir $(wildcard $(APPDIR)$(DELIM)*$(DELIM)Make.defs))
 BUILDIRS  := $(filter-out $(APPDIR)$(DELIM)import$(DELIM),$(BUILDIRS))
+CONFDIRS  := $(filter-out $(APPDIR)$(DELIM)builtin$(DELIM),$(BUILDIRS))
+CONFDIRS  := $(filter-out $(dir $(wildcard $(APPDIR)$(DELIM)*$(DELIM)Kconfig)),$(CONFDIRS))
 CLEANDIRS := $(dir $(wildcard $(APPDIR)$(DELIM)*$(DELIM)Makefile))

 # CONFIGURED_APPS is the application directories that should be built in
diff --git a/Makefile b/Makefile
index 91ef3a91..dff648bc 100644
--- a/Makefile
+++ b/Makefile
@@ -149,7 +149,7 @@ context_serialize:
 context: context_serialize

 Kconfig:
-   $(foreach SDIR, $(BUILDIRS), $(call MAKE_template,$(SDIR),preconfig))
+ $(foreach SDIR, $(CONFDIRS), $(call MAKE_template,$(SDIR),preconfig))
        $(Q) $(MKKCONFIG)

 preconfig: Kconfig

```

v01d, 2020-07-24:
Sorry if this is unrelated, but just in case it isn't and since you're investigating this part of the build system, I also always noticed that if you exit the menuconfig without saving you end up without a config.h until the next make, which I always found undesireable. I this is unrelated, I can open a separate issue.

Ouss4, 2020-07-24:
config.h is generated during context which is not run until pass1 or pass2.
https://github.com/apache/incubator-nuttx/blob/8792ade0fe50f125a4bc3b641cea7fd9f9baca9e/tools/Makefile.unix#L328
I don't think you'll get a config.h even if you saved.  You need to run make to have one.

closed 2020-07-25 by patacongo