apache/nuttx-apps #375: Build failure in apps/ for the 'ping' command

dosadi, 2020-08-30:
make[2]: Entering directory '/home/jon/controlix-nuttx/nuttx/apps/system/ping'
CC:  ping.c
gcc -c -Wall -Wstrict-prototypes -Wshadow -Wundef -O2 -fno-optimize-sibling-calls -fno-omit-frame-pointer -fno-crossjumping -fno-delete-null-pointer-checks -fPIC -fno-builtin -fno-stack-protector -mno-red-zone -mrdrnd -isystem "/home/jon/controlix-nuttx/nuttx/nuttx/include"   -pipe -I "/home/jon/controlix-nuttx/nuttx/apps/include" -Dmain  ping.c -o  ping.home.jon.controlix-nuttx.nuttx.apps.system.ping.o
<command-line>: error: expected identifier or ‘(’ before numeric constant
<command-line>: error: expected identifier or ‘(’ before numeric constant
ping.c:221:5: note: in expansion of macro ‘main’
  221 | int main(int argc, FAR char *argv[])
      |     ^~~~

Ouss4, 2020-09-01:
Are you using the latest code from the repo?  I just tested it and it's building fine.

> nuttx.nuttx.apps.system.ping.o
: error: expected identifier or ‘(’ before numeric constant
: error: expected identifier or ‘(’ before numeric constant

I'm not aware of any error like the above that was fixed recently, but that looks like something is wrong with Application.mk, do you have any changes there?

dosadi, 2020-09-01:
I am using the latest code from the repo, yes - as of yesterday.  I added CONFIG_SYSTEM_PING=y to my .config by hand, if that matters.

Ouss4, 2020-09-01:
Please don't edit the .config manually.  Select options from menuconfig and run make again.

dosadi, 2020-09-01:
I can't seem to find where in the huge list of config options that menuconfig provides that I can enable the 'ping' command in NSH.  Maybe you could help me?  Besides, I need to be able to hard-wire the .config for my application - I don't want either myself or the people who are going to have to build my project to need to run menuconfig in order to get the project to build.  I am hardwiring the options I need into tools/configure.sh.  Surely there is a set of .config variables that can correctly enable building of the ping command?  I found the variables for enabling 'ifconfig', 'ifup' and 'ifdown', so there if there are more .config variables I need to enable the ping command correctly (other then CONFIG_SYSTEM_PING), I need a way to find those....

acassis, 2020-09-01:
> I can't seem to find where in the huge list of config options that menuconfig provides that I can enable the 'ping' command in NSH. Maybe you could help me? Besides, I need to be able to hard-wire [...]

Hi @dosadi you can find it easily inside the menuconfig: hit "/" and type the symbol you are looking for i.e. SYSTEM_PING

btashton, 2020-09-01:
> I can't seem to find where in the huge list of config options that menuconfig provides that I can enable the 'ping' command in NSH. Maybe you could help me? Besides, I need to be able to hard-wire [...]

Also once you have a .config you are happy with, you can save this using the `make savedefconfig` this will generate a minimal defconfig that you can then share.

dosadi, 2020-09-01:
Thanks, @acassis, That helped - I hadn't enabled ICMP, so the menu option for ping didn't show up.  Once I did that, and tried again to 'make', I got a different error:

CC:  netlib_setifstatus.c
netlib_setifstatus.c: In function ‘netlib_ifup’:
netlib_setifstatus.c:83:27: error: ‘NETLIB_SOCK_TYPE’ undeclared (first use in this function); did you mean ‘NETLIB_SOCK_FAMILY’?
   83 |                           NETLIB_SOCK_TYPE, NETLIB_SOCK_PROTOCOL);
      |                           ^~~~~~~~~~~~~~~~
      |                           NETLIB_SOCK_FAMILY
netlib_setifstatus.c:83:27: note: each undeclared identifier is reported only once for each function it appears in
netlib_setifstatus.c: In function ‘netlib_ifdown’:
netlib_setifstatus.c:127:27: error: ‘NETLIB_SOCK_TYPE’ undeclared (first use in this function); did you mean ‘NETLIB_SOCK_FAMILY’?
  127 |                           NETLIB_SOCK_TYPE, NETLIB_SOCK_PROTOCOL);
      |                           ^~~~~~~~~~~~~~~~
      |                           NETLIB_SOCK_FAMILY
make[2]: *** [/home/jon/nuttx-test/apps/Application.mk:127: netlib_setifstatus.home.jon.nuttx-test.apps.netutils.netlib.o] Error 1

Ouss4, 2020-09-02:
You are missing other CONFIGs_.  What board are you using?  Did you take look at the configs that enable networking?
Here is a list:

```
boards/renesas/rx65n/rx65n-rsk2mb/configs/netnsh
boards/renesas/rx65n/rx65n-rsk1mb/configs/netnsh
boards/renesas/rx65n/rx65n-grrose/configs/netnsh
boards/arm/stm32/viewtool-stm32f107/configs/netnsh
boards/arm/stm32/stm32f4discovery/configs/netnsh
boards/arm/stm32/olimex-stm32-e407/configs/netnsh
boards/arm/stm32/nucleo-f429zi/configs/netnsh
boards/arm/kinetis/freedom-k66f/configs/netnsh
boards/arm/kinetis/twr-k64f120m/configs/netnsh
boards/arm/kinetis/freedom-k64f/configs/netnsh
boards/arm/sama5/sama5d2-xult/configs/netnsh
boards/arm/imxrt/imxrt1060-evk/configs/netnsh
boards/arm/imxrt/imxrt1050-evk/configs/netnsh
boards/arm/imxrt/imxrt1020-evk/configs/netnsh
boards/arm/samv7/same70-xplained/configs/netnsh
boards/arm/samv7/samv71-xult/configs/netnsh
boards/arm/lpc17xx_40xx/lincoln60/configs/netnsh
boards/arm/lpc54xx/lpcxpresso-lpc54628/configs/netnsh
boards/arm/stm32f7/nucleo-144/configs/f767-netnsh
boards/arm/stm32f7/stm32f746g-disco/configs/netnsh
boards/arm/stm32f7/stm32f769i-disco/configs/netnsh
boards/arm/lpc43xx/bambino-200e/configs/netnsh
boards/xtensa/esp32/esp32-core/configs/netnsh
```

acassis, 2020-09-02:
Hi @dosadi as @Ouss4 suggested it is a good idea to use an existing network configuration as base.

It appears you enabled NET_ICMP, but your forgot to enable NET_ICMP_SOCKET:

$ cd ../apps/
$ git grep NETLIB_SOCK_TYPE
$ vi include/netutils/netlib.h

#if NETLIB_SOCK_FAMILY == AF_INET
#  if defined(CONFIG_NET_UDP)
#    define NETLIB_SOCK_TYPE SOCK_DGRAM
#  elif defined(CONFIG_NET_TCP)
#   define NETLIB_SOCK_TYPE SOCK_STREAM
#  elif defined(CONFIG_NET_ICMP_SOCKET)   <----------------------------------- HERE
#   define NETLIB_SOCK_TYPE SOCK_DGRAM
#   undef NETLIB_SOCK_PROTOCOL
#   define NETLIB_SOCK_PROTOCOL IPPROTO_ICMP
#  endif

dosadi, 2020-09-02:
I am using:

* The standard config stuff from qemu-intel64:nsh

* My own set of hardwired CONFIG_* parameters, listed below:

CONFIG_LIBM=y
CONFIG_HAVE_CXX=y
CONFIG_LIBC_NETDB=y
CONFIG_NET=y
CONFIG_NSH_DISABLE_IFCONFIG=n
CONFIG_NSH_DISABLE_IFUPDOWN=n
CONFIG_NSOCKET_DESCRIPTORS=10
CONFIG_NET_TCP=y
CONFIG_NET_TCP_NPOLLWAITERS=1
CONFIG_IOB_NCHAINS=128
CONFIG_MM_IOB=y
CONFIG_IOB_NBUFFERS=1024
CONFIG_IOB_BUFSIZE=128
CONFIG_NET_TCP_WRITE_BUFFERS=y
CONFIG_NET_IPv4=y
CONFIG_NET_TCP_NWRBCHAINS=8
CONFIG_NET_ETHERNET=y
CONFIG_NET_SOCKOPTS=y
CONFIG_SYSTEM_PING=y
CONFIG_16550_REGINCR=1
CONFIG_16550_REGWIDTH=8

dosadi, 2020-09-02:
Success!  I found a couple more options I needed to turn on, and I got my ping application to show up and run in nsh.  Thanks to all who helped.  I am closing this bug.

closed 2020-09-02 by dosadi

acassis, 2020-09-02:
Kudos @dosadi