apache/nuttx-apps #398: make should traverse all path subcomponents of CONFIGURED_APPS

v01d, 2020-09-19:
I'm opening this issue since I tried to address some problem where an app was not being cleaned due to the current approach of detecting placeholder files not working (since it requires the placeholders to be present along the complete path to the app for it to work) in #397 but the problem requires a better solution.

To sum up, we can:
* recurse every subdirectory in `apps/`: slow and wasteful, not really necessary (brute force approach)
* recurse to directories containing placeholder files (such as .depend, etc.): this is prone to problems and as I mentioned fails in some cases
* go directly into each app directory from top-level Makefile: does not work since we need to for example clean generated Kconfig files in intermediate directories leading to an app dir
* **preferred** use CONFIGURED_APPS to build the list of all intermediate subdirectories we need to traverse

I'm trying the last approach but I couldn't get it fully to work. The first change is to have this in Make.defs:
```
CONFIGURED_SUBDIRS = $(sort $(foreach APP, $(CONFIGURED_APPS), $(CURDIR)$(DELIM)$(firstword $(subst /, ,$(subst $(CURDIR),,$(APP))))$(DELIM)))
```
This will take for example apps/examples/hello, apps/builtin, apps/wireless/bluetooth/btsak and, when standing on apps/ it will return examples, builtin, wireless. This works for the first level, but once you get into the second level, you should only consider apps for that subdirectory and this is where I'm not sure how to handle (eg: inside examples, do not consider the other apps to build the list of subdirectories).

patacongo, 2020-09-19:
>     * recurse every subdirectory in `apps/`: slow and wasteful, not really necessary (brute force approach)

originally, the system cleaned only CONFIGURED_APPS.  But that stranded object files after a reconfiguration.  In that case, object files in directories no longer in the configuration would be stranded forever.  Cleaning everything eliminated that problem.

v01d, 2020-09-19:
But now it is not cleaning *everything*, it is guided by the placeholder files and as I mentioned it is not enough.
The brute force would really clear *everything* but that is only reasonable if one always does make -j8 at least.

xiaoxiang781216, 2020-09-19:
List the related patch:
https://github.com/apache/incubator-nuttx-apps/pull/395
https://github.com/apache/incubator-nuttx/pull/1843

v01d, 2020-09-19:
Nevermind, I found that this was due to a "dirty" workspace: I was missing a .kconfig and since this is only regenerated after distclean it would not include this app to the cleandirs. I will close this since it is not actually a bug, but I think we should revisit this in the future, right now the build system is extremely complex and hard to maintain.

closed 2020-09-19 by v01d