apache/nuttx-apps #481: Elf loader sample app doesn't build

kernelgurumeditation, 2020-11-16:
tested with master

make[3]: Entering directory '/root/nuttx/apps/examples/elf'
make[4]: Entering directory '/root/nuttx/apps/examples/elf/tests'
make[5]: Entering directory '/root/nuttx/apps/examples/elf/tests/errno'
LD: errno.o
arm-none-eabi-ld: warning: cannot find entry symbol _start; defaulting to 00008000
errno.o: In function `main':
errno.c:(.text+0x2c): undefined reference to `fwrite'
errno.c:(.text+0x48): undefined reference to `fwrite'
errno.c:(.text+0x54): undefined reference to `fopen'
errno.c:(.text+0x80): undefined reference to `fprintf'
errno.c:(.text+0x88): undefined reference to `exit'
errno.c:(.text+0x98): undefined reference to `__errno'
errno.c:(.text+0xb0): undefined reference to `fprintf'
errno.c:(.text+0xc8): undefined reference to `_impure_ptr'
make[5]: *** [Makefile:70: errno] Error 1

btashton, 2020-11-19:
@kernelgurumeditation which configuration are you trying this with? There are some examples such as `spresense:elf` that I can confirm build.  You can find that config here `boards/arm/cxd56xx/spresense/configs/elf/defconfig`

kernelgurumeditation, 2020-11-19:
hi @btashton ,

thanks for the reply,

I'm trying with nucleo-h743zi/nsh and enabling the Elf support and the Elf loader sample App... should it not work?

I had a look at the spresense/configs/elf/defconfig file, and there is no option difference that would explain the failure... 

Is this an issue with the link scripts?

Thanks

btashton, 2020-11-19:
That is helpful, I have that board, I will try to do a functional test this evening.

patacongo, 2020-11-19:
There are changes needed to the nucleo-h743zi/scripts/Make.defs file as 
well.

See diff stm32h7/nucleo-h743zi/scripts/Make.defs 
cxd56xx/spresense/scripts/Make.defs

Here are some necessary additions:

72a79,103
 > # Loadable module definitions
 >
 > CMODULEFLAGS = $(CFLAGS) -mlong-calls # --target1-abs
 >
 > LDMODULEFLAGS = -r -e module_initialize
 > ifeq ($(CONFIG_CYGWIN_WINTOOL),y)
 >   LDMODULEFLAGS += -T "${shell cygpath -w 
$(TOPDIR)/libs/libc/modlib/gnu-elf.ld}"
 > else
 >   LDMODULEFLAGS += -T $(TOPDIR)/libs/libc/modlib/gnu-elf.ld
 > endif
 >
 > # ELF module definitions
 >
 > CELFFLAGS = $(CFLAGS)
 > CXXELFFLAGS = $(CXXFLAGS)
 >
 > LDELFFLAGS = -r -e main
 > ifeq ($(CONFIG_CYGWIN_WINTOOL),y)
 >   LDELFFLAGS += -T "${shell cygpath -w 
$(BOARD_DIR)$(DELIM)scripts$(DELIM)gnu-elf.ld}"
 > else
 >   LDELFFLAGS += -T $(BOARD_DIR)$(DELIM)scripts$(DELIM)gnu-elf.ld
 > endif

The latter ELF definitions are certainly necessary.

xiaoxiang781216, 2020-11-20:
But, it's the wrong direction to duplicate the common setting to individual Make.defs. I have saw many people complain some advanced features don't work(e.g. elf, c++...), just because Make.defs doesn't have the necessary options. We need fix the problem for all board, otherwise people will complain again and again.

kernelgurumeditation, 2020-11-20:
Well,  talking as the new Guy..... my main issue is not knowing what is
expected to work.
This is an RTOS, not windows, typically the ppl who search/work on this is
not expecting all to work... but we would like to know, what is expected to
work...

What is very interesting for me regarding NuttX is being a Posixish system
and having a   shell makes it very comfortable for a Unix guy like me, and
it's really amazing what this little OS can do :)

PS: I tried to replicate the Make.defs settings and it didn't fix it...

btashton, 2020-11-20:
@xiaoxiang781216 I agree that we should move more of this code into a single place, but for now I added a fully functional configuration for his board.  This is also helpful for me because I do not have the other boards that use this configuration so I can better test this in the future.

I also want to work on getting these features added into the new documentation as well.

kernelgurumeditation, 2020-11-20:
@btashton, thanks for fixing it.

PS: I fixed the USB HOST OTG port for this board, if you want the fix I can send it.

closed 2020-11-20 by kernelgurumeditation

btashton, 2020-11-20:
Sure I'm happy to review it. I really want to pull some of the common board logic out so we don't have to keep copy pasting it for every variant, but that will take some time.

kernelgurumeditation, 2020-11-20:
Ok, 
it's at    https://github.com/apache/incubator-nuttx/pull/2353
Not pretty but works, and can be useful to someone :)