apache/nuttx-apps #539: "make distclean" leaves a few auto generated Kconfig

yamt, 2020-12-25:
```
wireless/bluetooth/Kconfig:# This file is autogenerated, do not edit.
wireless/ieee802154/Kconfig:# This file is autogenerated, do not edit.
```

Ouss4, 2020-12-25:
@yamt weren't these generated in a previous build?  By previous I mean several months ago before some changes were made to the build system, mainly Application.mk and Directory.mk.
Or maybe a manual clean was performed? maybe git clean -xdf?

The clean now doesn't recursively traverse all the directories in apps. It looks for .kconfig or .depend files.  If those files aren't there that directory won't be cleaned.

I just tried one config (esp32-devkitc:wapi) and the distclean deleted all the autogenerated files.

yamt, 2020-12-25:
@Ouss4 thank you. probably it's the case as the tree was quite long-lived. i'll close this for now until i have a chance to confirm.

closed 2020-12-25 by yamt