apache/nuttx-apps #592: nimble sim configuration is breaking CI beacuse of pointer-to-int-cast error

btashton, 2021-02-16:
@v01d  Looks like the nimble app is pointed at master not a commit or tag so it just started breaking builds like this:
I commented on the PR that caused this https://github.com/apache/mynewt-nimble/pull/872
I'll open a PR to pin against the previous commit for now.
```
Configuration/Tool: sim/nimble
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Building NuttX...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   128    0   128    0     0    761      0 --:--:-- --:--:-- --:--:--   761

100 1514k    0 1514k    0     0  2435k      0 --:--:-- --:--:-- --:--:-- 2435k
/github/workspace/sources/apps/wireless/bluetooth/nimble/mynewt-nimble/porting/nimble/src/os_mbuf.c: In function 'os_mbuf_get':
Error: /github/workspace/sources/apps/wireless/bluetooth/nimble/mynewt-nimble/porting/nimble/src/os_mbuf.c:247:46: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  247 |     os_trace_api_u32x2(OS_TRACE_ID_MBUF_GET, (uint32_t)omp,
      |                                              ^
Error: /github/workspace/sources/apps/wireless/bluetooth/nimble/mynewt-nimble/porting/nimble/src/os_mbuf.c:268:48: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  268 |     os_trace_api_ret_u32(OS_TRACE_ID_MBUF_GET, (uint32_t)om);
```

v01d, 2021-02-16:
Sorry for the issue, don't know why I didn't do it sooner. Thanks for the fix.
I think that this kind of errors will have to be dealt with disabling certainly warnings when building nimble. I think they use a much more loose setting than us.

btashton, 2021-02-16:
I think I would like to avoid it if possible, they already have a PR up to fix it. In this case it actually is calling out a bug. I suspect they did not see it because there CI seems to only target 32bit and not 64, like the sim.

closed 2021-03-28 by xiaoxiang781216