apache/nuttx-apps #692: libcanard refers to nonexistant sources

v01d, 2021-04-22:
During CMake migration I found it refers to sources not present on libcanard repo: canard_nuttx.c and .h
Maybe this was actually using a fork?

Ouss4, 2021-04-23:
The script downloads an old version that had those files: https://github.com/UAVCAN/libcanard/tree/5ad65c6a4efda60cda7a8f0512da0f465822bbb8/drivers/nuttx

davids5, 2021-04-23:
If it will build with the `legacy-v0` branch that is what should be used. If not we should work on making it work `legacy-v0`.

v01d, 2021-04-23:
> The script downloads an old version that had those files: https://github.com/UAVCAN/libcanard/tree/5ad65c6a4efda60cda7a8f0512da0f465822bbb8/drivers/nuttx

You're right. I didn't realize it was in a subdirectory.

closed 2021-04-23 by v01d