apache/nuttx-apps #831: when i make for esp32c3 ,run make

anjihao, 2021-08-15:


anjihao, 2021-08-15:
ubuntu20.4 for wsl
i use `sudo apt-get install gcc-riscv64-unknown-elf`
and `cd nuttx`
`./tools/configure.sh eps32c3-devkit:gpio`
`make -16`

anjihao, 2021-08-15:
```
make[2]: Leaving directory '/home/ajh/work/apps'
make[1]: Leaving directory '/home/ajh/work/apps'
IN: /home/ajh/work/apps/libapps.a -> staging/libapps.a
make[1]: Entering directory '/home/ajh/work/nuttx/arch/risc-v/src'
make[2]: Entering directory '/home/ajh/work/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/src'
make[2]: 'libboard.a' is up to date.
make[2]: Leaving directory '/home/ajh/work/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/src'
LD: nuttx
riscv64-unknown-elf-ld: cannot find -lm
make[1]: *** [Makefile:145: nuttx] Error 1
make[1]: Leaving directory '/home/ajh/work/nuttx/arch/risc-v/src'
make: *** [tools/Makefile.unix:422: nuttx] Error 
```

cannot find -lm? how to fix it?

anjihao, 2021-08-15:
,i use -lgcc to repace $(EXTRA_LIBS)
`./tools/configure.sh esp32c3-devkit:gpio`
`make -j16`
```
AR (add): libapps.a    dummy.c.home.ajh.work.apps.platform.o   
make[3]: Leaving directory '/home/ajh/work/apps/platform'
make[2]: Leaving directory '/home/ajh/work/apps'
make[1]: Leaving directory '/home/ajh/work/apps'
IN: /home/ajh/work/apps/libapps.a -> staging/libapps.a
make[1]: Entering directory '/home/ajh/work/nuttx/arch/risc-v/src'
make[2]: Entering directory '/home/ajh/work/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/src'
make[2]: 'libboard.a' is up to date.
make[2]: Leaving directory '/home/ajh/work/nuttx/boards/risc-v/esp32c3/esp32c3-devkit/src'
LD: nuttx
make[1]: Leaving directory '/home/ajh/work/nuttx/arch/risc-v/src'
CP: nuttx.hex
CP: nuttx.bin
MKIMAGE: ESP32-C3 binary
esptool.py -c esp32c3 elf2image -fs 4MB -fm dio -ff 40m -o nuttx.bin nuttx
esptool.py v3.1
Generated: nuttx.bin (ESP32-C3 compatible)
```

anjihao, 2021-08-15:
and `./tools/configure.sh esp32c3-devkit:sta_softap`
`sudo make -j16`

```
make[3]: Leaving directory '/home/ajh/work/apps/netutils/netlib'
make[3]: Nothing to be done for 'register'.
make[3]: Leaving directory '/home/ajh/work/apps/platform'
make[3]: Nothing to be done for 'register'.
make[3]: Leaving directory '/home/ajh/work/apps/netutils/dhcpd'
make[3]: Nothing to be done for 'register'.
make[3]: Leaving directory '/home/ajh/work/apps/wireless/wapi'
make[2]: Leaving directory '/home/ajh/work/apps'
make[1]: Leaving directory '/home/ajh/work/apps'
make[1]: Entering directory '/home/ajh/work/nuttx/arch/risc-v/src'
Unpacking: ESP Wireless Drivers
mv: cannot move 'chip/esp-wireless-drivers-3rdparty-024420dcd05706242b781fe68064c6830936ef86' to 'chip/esp-wireless-drivers-3rdparty': Permission denied
make[1]: *** [chip/Make.defs:190: chip/esp-wireless-drivers-3rdparty] Error 1
make[1]: Leaving directory '/home/ajh/work/nuttx/arch/risc-v/src'
make: *** [tools/Makefile.unix:343: context] Error 2
```

acassis, 2021-08-15:
Hi @anjihao could you please try it: https://acassis.wordpress.com/2021/01/29/compiling-nuttx-to-esp32-c3-no-public-yet/

anjihao, 2021-08-15:
> Hi @anjihao could you please try it: https://acassis.wordpress.com/2021/01/29/compiling-nuttx-to-esp32-c3-no-public-yet/

think you ,fix it!!!!

closed 2021-08-15 by anjihao