apache/nuttx-apps #834: Trying to run the Timer example

eden-desta, 2021-08-17:
Hello! I am trying to run the timer example following the steps provided here: https://nuttx.apache.org/docs/latest/components/drivers/character/timer.html

The few things that  I do that I do not see in the steps, is also selecting a peripheral timer, from the STM32 peripheral tag. 

I get the error `Failed to open /dev/timer0: 2` when running the example. A quick `ls` of the `/dev` shows the following:
```
console
 null
 ttyS0
```
Why isnt `/dev/timer0` part of this list. 

++ I am using a Nucleo-H743ZI2 board with pretty much nothing else running other than the simple timer example. From the link provided i followed the second set of steps via the menuconfig

saramonteiro, 2021-08-17:
Hi @eden-desta!

You're not seeing it on your `/dev` dir because the board bringup function of this board is not calling the function that register the timer drivers.
See your board bringup: https://github.com/apache/incubator-nuttx/blob/master/boards/arm/stm32h7/nucleo-h743zi2/src/stm32_bringup.c
Take this bringup on ESP32 as an example: https://github.com/apache/incubator-nuttx/blob/master/boards/xtensa/esp32/esp32-devkitc/src/esp32_bringup.c#L227-L249
In your case you should call `stm32_timer_initialize` to register it.

eden-desta, 2021-08-18:
Hey @saramonteiro, thanks for the response!
So I dont kno if there is something wrong with my process here, but I am still seeing the same issue. 
This is now in my `stm32_bringup.c`:
```
#ifdef CONFIG_TIMER
  ret = stm32_timer_initialize("/dev/timer0", 0);
  if (ret < 0) {
    syslog(LOG_ERR, "ERROR: Failed to initialize timer driver: %d\n", ret);
  }
#endif
```

I have included the header file:
```
#ifdef CONFIG_TIMER
#include <stm32_tim.h>
#endif
```

Via the `menuconfig`
[image]
[image]
[image]

But i still unfortunately get:
[image]

eden-desta, 2021-08-18:
Hi @saramonteiro thank you for your response! I appreciate you!

Okay, so I have made some updates to the `stm32_bringup.c` file:
```
#ifdef CONFIG_TIMER
#include <stm32_tim.h>
#endif

....

#ifdef CONFIG_TIMER
  ret = stm32_timer_initialize("/dev/timer0", 0);
  if (ret < 0) {
    syslog(LOG_ERR, "ERROR: Failed to initialize timer driver: %d\n", ret);
  }
#endif
```

And i get the following error:
```
/home/eden/nuttx/nuttx/boards/arm/stm32h7/nucleo-h743zi2/src/stm32_bringup.c:185: undefined reference to `stm32_timer_initialize'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:156: nuttx] Error 1
make[1]: Leaving directory '/home/eden/nuttx/nuttx/arch/arm/src'
make: *** [tools/Makefile.unix:422: nuttx] Error 2
```

I noticed that the `esp32_bringup.c` includes `#include <esp32_tim_lowerhalf.h>` which is not available for the stm32 boards. So in an effort to see if perhaps this can be something I can work around, i imitated that `esp32_tim_lowerhalf.h` file and made an `stm32_tim_lowerhalf.h` file that I included in the `bringup` instead. I removed the `stm32_timer_initialize` function from the `stm32_tim.h` file and moved it to the new `lowerhalf`. When i built it, it said that the `stm32_tim_lowerhalf.h` files does not exist. 

So stuck on two avenues with a little bit of uncertainty which way is the right way to go about moving forward