apache/nuttx-apps #835: STM32H7 does not have the a functioning Timer driver

eden-desta, 2021-08-18:
Related to https://github.com/apache/incubator-nuttx-apps/issues/834

I was trying to determine a way to use the timer with my NUCLEOH743ZI2 board. Following the guidelines provided in the above issue, I was still experiencing some problems. I did a deep dive into the what the ultimate cause of the problem was. This is as follows:

When navigating to `nuttx -> arc -> arm -> src -> stm32` This folder contains a `stm32_tim_lowerhalf.c` file that defines the function `stm32_timer_initialize` which we need in order to get any timer up and operating.  However in the `stm32h7` folder found in the same location as the `stm32` folder I noticed that the `stm32_tim.h` file defines it, but the `stm32_tim.c` file does not integrate it. Additionally, the `stm32_tim_lowerhalf.c` file does not exist. 

Ultimately, this cause the `startup.c` file to fail on calling the `stm32_timer_initialize` function as it does not think that it exists. I am unsure how to go about using timers with the `H7` board. 

Any thoughts, or feedback would be great! Thank you for your time!

saramonteiro, 2021-08-18:
Yes. It seems the timer is still not completed on `smt32h7`.
The low level functions are implementes in `nuttx/arch/arm/src/stm32h7/stm32_tim.c` but it's missing to implement the lowerhalf.
The `stm32_timer_initialize` is only declared but not defined.
You should define it, and the other lowerhalf operations, on the lowerhalf file.

acassis, 2021-08-18:
Hi @eden-desta the stm32h7 is a new port and is missing some pieces of code. The good news is H7 is very similar to F7, so I suggest to to use the stm32_tim_lowerhalf.c from stm32f7 and double check the registers address in the datasheet. It is pretty easy to do. If you need some help, please subscribe to NuttX mailing list, see here: https://nuttx.apache.org/community/
Also, this weekend will have the NuttX Online Workshop, case you are not aware of it. Felipe will talk about microROS on ESP32.

eden-desta, 2021-08-18:
Amazing, thank you both @acassis and @saramonteiro! I will try and get this going asap. 

@acassis just confirming this is the datasheet i should refer to: https://www.st.com/resource/en/datasheet/stm32h753ii.pdf

acassis, 2021-08-18:
No, I said datasheet, but the register are describe in the Reference Manual: https://www.st.com/en/microcontrollers-microprocessors/stm32h753ii.html#documentation

acassis, 2021-08-18:
Don't get afraid of its more than 3000 pages, you just need to look at timer related registers page

eden-desta, 2021-08-18:
Amazing found it! Thank you!