apache/nuttx-apps #858: tools/mksymtab.sh does not work properly in Windows MSYS

baggio63446333, 2021-09-28:
https://github.com/apache/incubator-nuttx-apps/blob/cb1b8e10db9fdf15e7dc1b4ccb5c2024d11a7713/tools/mksymtab.sh#L45

In `tools/mksymtab.sh`, the file permission is used to get the execlist from https://github.com/apache/incubator-nuttx-apps/pull/271, but in MSYS environment, the file permission cannot be changed by `chmod +x` of `Application.mk`. As a result, `symtab_apps.c` cannot be generated correctly. In versions older than NuttX 9.1.0, there was no permission decision, so it worked as intended.

For example, if I delete `-perm -a=x`, it also works correctly in the MSYS environment, but I don't know if there are any side effects or not.

Anyone have a solution? or may I delete `-perm -a=x`?

xiaoxiang781216, 2021-09-30:
@baggio63446333 do you know MSYS fail to run "chmod +x"?

patacongo, 2021-09-30:
This is a property of Windows file systems.  The file systems do not retain a executable bit.  In Cygwin (and MSYS which is a Cygwin variant), this property is determined by the file extension.  If the file ends in .exe, for example, the file will be reported as executable.  See https://cygwin.com/cygwin-ug-net/using-filemodes.html

patacongo, 2021-09-30:
Details of the behaviors differ with file system and are not the same for Cygwin and MSYS2:

Cygwin on NTFS:

    $ touch testfile

    $ ls -l testfile
    -rwxrw-r--+ 1 spuda spuda 0 Sep 30 09:10 testfile

    $ chmod +x testfile

    $ ls -l testfile
    -rwxrwxr-x+ 1 spuda spuda 0 Sep 30 09:10 testfile

MSYS2 on NTFS:

    $ ls -l testfile
    -rw-r--r-- 1 spuda spuda 0 Sep 30 09:10 testfile

    $ rm testfile

    $ touch anotherfile

    $ ls -l anotherfile
    -rw-r--r-- 1 spuda spuda 0 Sep 30 09:16 anotherfile

    $ chmod +x anotherfile

    $ ls -l anotherfile
    -rw-r--r-- 1 spuda spuda 0 Sep 30 09:16 anotherfile

patacongo, 2021-09-30:
Interesting but off-topic:  There are instructions for setting up MSYS2 in github CI here: https://www.msys2.org/docs/ci/

closed 2021-10-05 by xiaoxiang781216