apache/nuttx-apps #886: Compiling for the sim:nsh configuration fails

cocoacrumbs, 2021-11-07:
I followed the simple instructions for compiling the `sim:nsh` configuration (https://nuttx.apache.org/docs/latest/guides/simulator.html). 
Near the end of the compilation, I get a compilation failure:

```
/home/koen/Documents/Projects/NuttX/sim/master/apps/nshlib/nsh_romfsetc.c:89: undefined reference to `romfs_img_len'
/usr/bin/ld: /home/koen/Documents/Projects/NuttX/sim/master/apps/nshlib/nsh_romfsetc.c:91: undefined reference to `romfs_img'
```

This is the piece of code where the references to `romfs_img_len` and `romfs_img` are made (`CONFIG_NSH_CROMFSETC` is not set in this configuration):

```
int nsh_romfsetc(void)
{
  int  ret;

#ifndef CONFIG_NSH_CROMFSETC
  struct boardioc_romdisk_s desc;

  /* Create a ROM disk for the /etc filesystem */

  desc.minor    = CONFIG_NSH_ROMFSDEVNO;     /* Minor device number of the RAM disk. */
  desc.nsectors = NSECTORS(romfs_img_len);   /* The number of sectors in the RAM disk */
  desc.sectsize = CONFIG_NSH_ROMFSSECTSIZE;  /* The size of one sector in bytes */
  desc.image    = romfs_img;                 /* File system image */

  ret = boardctl(BOARDIOC_ROMDISK, (uintptr_t)&desc);
  if (ret < 0)
    {
      ferr("ERROR: boardctl(BOARDIOC_ROMDISK) failed: %d\n", -ret);
      return ERROR;
    }
#endif
```

I got the code compiled by manually including the `nsh_romfsimg.h` header but that's not the proper solution of course.

The comments are confusing me as well. This is for a **ROM** disk, but the comments in the code speak of a **RAM** disk?

My feeling is that the `nsh_romfsimg.h` header should be included if `CONFIG_NSH_CROMFSETC` is not defined but I'm a complete newbie and I'm probably wrong in this (maybe the `sim:nsh` configuration itself is faulty?).

jcursiolf, 2022-03-15:
If it helps anyone, I had the same problem when using branch `releases/10.2`. After creating a branch from the tag `nuttx-10.2.0` it worked (`git checkout nuttx-10.2.0 -b tag-10.2`, for example).
I haven't looked into the differences between the tag `nuttx-10.2.0` and the branch `releases/10.2`, but there shouldn't be, ideally.

casaroli, 2026-11-08:
Possibly fixed by #2256 and apache/nuttx#11498.

closed 2026-11-08 by casaroli