apache/nuttx-apps #947: Compile MCUBOOT failed using mcuboot_agent config on esp32-wrover-kit

PierrePeng, 2021-12-30:
Hi @gustavonihei

I encouter to obstacle when I compile mcuboot using command as bellow. It seems that the configuration of compiler on NuttX uses C89 as default
`./tools/configure.sh esp32-wrover-kit:mcuboot_agent && make -j16`
[image]

After I move the declarations out off for loop, I got multiple definition errors.
[image]

gustavonihei, 2021-12-30:
Hi @PierrePeng.

> ./tools/configure.sh esp32-wrover-kit:mcuboot_agent && make -j16

The `esp32-devkitc:mcuboot_agent`, which is essentially equal to `esp32-wrover-kit:mcuboot_agent`, is part of the `incubator-nuttx` CI and it builds successfully using the `xtensa-esp32-elf-gcc` compiler from Espressif, which is the one assumed by the defconfig.
Could you please inform the version of the toolchain used to reproduce the issue?

If by any chance you modified the base `mcuboot_agent` defconfig, could you please share those changes?

PierrePeng, 2021-12-31:
Hi @gustavonihei ,

Thanks for your reply! 
The version of `xtensa-esp32-elf-gcc` is shown below.  And I didn't modify the base `mcuboot_agent` deconfig.

[image]

PierrePeng, 2021-12-31:
@gustavonihei 

Now, I can compile successfully after I update the compiler to 8.4.0. 

Thanks!!!

closed 2021-12-31 by PierrePeng