apache/nuttx-apps #978: Delete platform folder?

AlanRosenthal, 2022-01-24:
Hi,

I'm doing some cleanup work on the Makefiles in incubator-nuttx. See https://github.com/apache/incubator-nuttx/issues/5205 for more details.

I'm trying to figure out exactly what the `platform` folder is being used for.

It currently only has one non-dummy entry: `platform/mikroe-stm32f4`.

Q1: Is this leftover from the nuttx-app split?
Q2: Should `platform/mikroe-stm32f4` be moved to another location?
`platform/mikroe-stm32f4/mikroe_configdata.c` implements `platform_setconfig` and `platform_getconfig`. The other functions that implement that API are defined in `graphics/nxwm/src/ccalibration.cxx`, `graphics/nxwm/src/nxwm_main.cxx`, `graphics/twm4nx/apps/ccalibration.cxx`. Maybe the correcct location is under `graphics`.

Also, it looks like most of the implementation of `platform/mikroe-stm32f4/mikroe_configdata.c` are guarded by `CONFIG_MIKROE_STM32F4_CONFIGDATA_FS` and `CONFIG_MIKROE_STM32F4_CONFIGDATA_PART`. I couldn't find any defconfig where those were set to `y`.

Q3: Is we move/delete `mikroe-stm32f4`, should we also delete the `platform` folder? At first look it doesn't make sense for apps to have a platform specific code.

patacongo, 2022-01-24:
The apps/platform directory was created to support platform-specific, i.e., board-specific, applications.  In the kernel build mode, the files under boards/ are built into the kernel and run in supervisor mode.  So if you want user-mode, board-specific code there is only apps/platform.

That facility has not been used much however.  But there really is no other place for board-specific,  user code.

From: Alan Rosenthal
Sent: Sunday, January 23, 2022 8:32 PM
To: apache/incubator-nuttx-apps
Cc: Subscribed
Subject: [apache/incubator-nuttx-apps] Delete platform folder? (Issue #978)

Hi,
I'm doing some cleanup work on the Makefiles in incubator-nuttx. See apache/incubator-nuttx#5205 for more details.
I'm trying to figure out exactly what the platform folder is being used for.
It currently only has one non-dummy entry: platform/mikroe-stm32f4.
Q1: Is this leftover from the nuttx-app split?
Q2: Should platform/mikroe-stm32f4 be moved to another location?
platform/mikroe-stm32f4/mikroe_configdata.c implements platform_setconfig and platform_getconfig. The other functions that implement that API are defined in graphics/nxwm/src/ccalibration.cxx, graphics/nxwm/src/nxwm_main.cxx, graphics/twm4nx/apps/ccalibration.cxx. Maybe the correcct location is under graphics.
Also, it looks like most of the implementation of platform/mikroe-stm32f4/mikroe_configdata.c are guarded by CONFIG_MIKROE_STM32F4_CONFIGDATA_FS and CONFIG_MIKROE_STM32F4_CONFIGDATA_PART. I couldn't find any defconfig where those were set to y.
Q3: Is we move/delete mikroe-stm32f4, should we also delete the platform folder? At first look it doesn't make sense for apps to have a platform specific code.

xiaoxiang781216, 2022-01-24:
How about we add support libboard.a and libkboard.a pair just like what we have done for arch(libarch.a and libkarch.a)? So, all board specific logic can hold in one place regardless whether it's for kernel space or user space.