apache/nuttx-apps #994: i2c-tools is not able to detect pressure sensor MS5611

acassis, 2022-02-02:
I decide to report it to prevent forgetting this issue. Although the device is not detected writing to it working correctly:
```
nsh> i2c set -a 77 1e
WROTE Bus: 0 Addr: 77 Subaddr: 00 Value: 1e
nsh> i2c set -a 77 a0
WROTE Bus: 0 Addr: 77 Subaddr: 00 Value: a0
nsh> i2c dump -a 77 2
DUMP Bus: 0 Addr: 77 Subaddr: --
00000000 08 04
```

linguini1, 2024-07-16:
I would like to second this, and I also noticed it cannot detect the SHT4X sensor as well.

I have a possible change to the i2c tool which may ameliorate this: I have created an I2C scanner before and in my experience, all the sensors I've worked with will respond to a single byte write of `0x00` to their address.

acassis, 2024-07-16:
Hi @linguini1 nice suggestion! Do you have a RaspberryPi board with Linux? I think it should be nice to see if it detects the sensor and see what it sends in the bus.

Probably it is doing as you suggested, but please use a logic analyzer to confirm. When we could use same approach.

linguini1, 2024-07-16:
> Hi @linguini1 nice suggestion! Do you have a RaspberryPi board with Linux? I think it should be nice to see if it detects the sensor and see what it sends in the bus.
> 
> Probably it is doing as you suggested, but please use a logic analyzer to confirm. When we could use same approach.

I have a Pi Pico and Pi 4 available to me, neither with Linux on it currently. I do have access to a logic analyzer in my university's lab. Just clarify, you want me to verify what the sends on the bus in response to the write request?

acassis, 2024-07-16:
> I have a Pi Pico and Pi 4 available to me, neither with Linux on it currently. I do have access to a logic analyzer in my university's lab. Just clarify, you want me to verify what the sends on the [...]

Yes, my suggest if verify what Linux i2ctools send over I2C bus (remember: you cannot look the source code and implement if for NuttX, GPL License and Apache are incompatible)

linguini1, 2024-07-16:
@acassis It seems from my poking around online that the correct approach would be to just send a start header (the address header) and check for an ACK response.

[See this example for instance](https://gms.tf/i2c-scanner.html#i2c-address-scanning)

Neither a read or write packet is required, just the address header. 0 bytes transmitted outside of that. I can try poking around with this later today.

linguini1, 2024-07-16:
I managed to get proper detection for all the sensors I have access to using the following logic:

```c
int main(int argc, FAR char *argv[]) {

  int err;
  int i2c = open("/dev/i2c0", O_RDWR);
  if (i2c < 0) {
    fprintf(stderr, "Couldn't open /dev/i2c0: %d\n", errno);
    return 1;
  }

  uint8_t contents = 0;
  for (unsigned int i = 0; i < 128; i++) {

    struct i2c_transfer_s i2c_transfer;
    struct i2c_msg_s msg = {
        .frequency = I2C_SPEED_FAST,
        .addr = i,
        .flags = 0,
        .buffer = &contents,
        .length = 0,
    };
    i2c_transfer.msgv = &msg;
    i2c_transfer.msgc = 1;

    err = ioctl(i2c, I2CIOC_TRANSFER, (unsigned long)(&i2c_transfer));
    if (!err) {
      printf("%02x\n", i);
    }
  }

  return 0;
}
```
Because the message length is set to 0, I would expect that the I2C driver is just sending the address header, but I will have to wait until I have access to a logic analyzer to confirm.

acassis, 2024-07-16:
@linguini1 I suggest you to use I2C_SPEED_STANDARD because some old device will not support 400KHz, and all devices will accept this frequency.

acassis, 2024-07-16:
@linguini1 could you please submit a patch to fix apps/system/i2c/i2c_dump.c ?

Modify i2ctool_dump() to do a single transfer (suggestion: maybe you could add a parameter option to support to "legacy" way)

acassis, 2024-07-16:
@linguini1 whoa... actually looking the code I see now that there is an option that just send a single msg:

```
  else
    {
      /* no register index "-r" has been specified so
       * we do a pure read (no write of index)
       */

      msg[0].frequency = i2ctool->freq;
      msg[0].addr      = i2ctool->addr;
      msg[0].flags     = I2C_M_READ;

      msg[0].buffer = buf;
      msg[0].length = nbytes;

      ret = i2cdev_transfer(fd, msg, 1);
    }
```

I missed that option: could you try passing the "-r" parameter during the scan?

linguini1, 2024-07-17:
@acassis No luck with the `-r <register address>` option. This is the output I see:

```c
nsh> i2c dev -r 00 00 77
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- 42 -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- 6b -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
nsh> 
```
`42`, `50` and `6b` are the sensors that always appear with the scanner. I should also be seeing `44` for the SHT41 and `76` and `77` for the MS5611 sensors.

acassis, 2024-07-17:
Thank you @linguini1, so go ahead and do your modification.

About my suggestion to add retro-compatibilty, let to ignore it and make things simple.

PR #2447 (2024-07-20): `i2ctool dev` command now uses an empty write header to scan the I2C bus
## Summary

The `i2ctool dev` command now supports an additional command-line flag, `-z`, to the `dev` command. This flag will use a zero-byte write request over I2C as the scan instead of the default 1 byte read. This sometimes allows for discovery of more devices, such as the SHT41 temperature and humidity sensor, and the MS5611 barometric pressure sensor.

This PR closes #994.

## Impact

This should save some developers time debugging or wondering why their devices didn't show up (and trying to wrongly diagnose a hard [... cut]
closed 2024-07-20 by acassis