apache/nuttx-apps #1058: LIBPATHS and EXTRA_LIBS  appending not work for app's Makefile

dangbinghoo, 2022-03-11:
if I put 

``` Makefile
EXTRA_LIBPATHS += -L $(PWD)
LIBPATHS += $(PWD)
EXTRA_LIBS += -lnimcache
```
it seems, when linking the final binary,  these flags were not appended to  LD.

and CFLAGS += also won't work, IS there any way to achieve this ?

xiaoxiang781216, 2022-03-11:
Yes, nuttx/Makefile invoke but not include apps/Makefile. So the change to EXTRA_LIBPATHS/EXTRA_LIBS can't propagate to nuttx/Makefile.

xiaoxiang781216, 2022-03-11:
It is welcome to provide a solution fix this problem.

dangbinghoo, 2022-03-11:
> 

is there any nuttx talking groups or what exists? 
PS: I'm from CN, QQ group is ok.

xiaoxiang781216, 2022-03-11:
Yes, here: https://nuttx.apache.org/community/

xiaoxiang781216, 2024-10-07:
@dangbinghoo Fixed by https://github.com/apache/nuttx/pull/11202

closed 2024-10-07 by xiaoxiang781216