apache/nuttx-apps #1153: The right way to redirect syslog to telnet?

XinStellaris, 2022-04-22:
Here is the background:
I have a dev-board which has limited number of pins. So I have to disable  CONFIG_NSH_CONSOLE and use all the pins for other purposes.  I have disabled it and booted the system, the system runs and gives the expected results(can be examined by detecting PWM or other peripheral output).
However, it is still painful without log output, so I want to redirect all syslog output to telnet.

After some thinking, I think one solution is:
1. Create a syslog channel when telnet_session is created. The code may be something like:
int syslog_telnet_channel(FAR struct telnet_session_s *session)
{
  g_syslog_telnet_channel = syslog_dev_initialize(session->ts_devpath,    // will be the same name as telnet device, 
                                               OPEN_FLAGS, OPEN_MODE);
  if (g_syslog_telnet_channel == NULL)
    {
      return -ENOMEM;
    }

  return syslog_channel(g_syslog_telnet_channel );
}
2. Also need to add lock mechanism in telnet_write, it will be called from different task now

Is this the right way?

xiaoxiang781216, 2022-04-23:
> Here is the background: I have a dev-board which has limited number of pins. So I have to disable CONFIG_NSH_CONSOLE and use all the pins for other purposes. I have disabled it and booted the [...]

This solution can't handle the log from interrupt or panic. Why not use ramlog?

pkarashchenko, 2022-04-23:
>    1. Also need to add lock mechanism in telnet_write, it will be called

I think for telnet log it should be enough just to
"syslog_file_channel(session->ts_devpath)", but you will need to configure
"CONFIG_SYSLOG_FILE_ROTATIONS=0" and "CONFIG_SYSLOG_FILE_SEPARATE=n". There
will be no need for additional lock in telnet_write because syslog_dev
layer already contain "ret = syslog_dev_takesem(syslog_dev);" and should be
sufficient to be used with multiple tasks. But the messages output may be
still messed up because upper layer does not lock at message boundary, but
only ensures that lower level writes are protected. But still that is again
the same as with UART logging.

And also I would agree that ramlog is a good alternative.

XinStellaris, 2022-04-24:
Yes, ramlog is an alternative. 
If my solution cant deal with interrupt or panic log. Then ramlog is better.

closed 2022-04-24 by XinStellaris