apache/nuttx-apps #1322: LVGL v8 version upgrade

FASTSHIFT, 2022-09-24:
[LVGL v8](https://github.com/lvgl/lvgl/releases/tag/v8.3.1)  has been released for a while, and we should consider upgrading `graphics/lvgl` to that version.

Modifications that may be involved in an upgrade:
1. Delete LVGL v7-based Makefile, Kconfig configuration file, and related dependency files
2. Add new and better porting based on LVGL v8, including fb, lcd, button, etc.
3. Modify `examples/lvgldemo`, use the demo of LVGL v8, and remove the porting function here

Is there anything else?

acassis, 2022-09-27:
@FASTSHIFT yes, please helps us if you can, adding v8 support will allow us to use SquareLine Studio to create interfaces very easily!

xiaoxiang781216, 2022-09-28:
@lupyuen we plan to upgrade lvgl from v7 to v8, do you have any concern?

closed 2023-01-01 by acassis

lupyuen, 2023-01-01:
Tested OK on PinePhone. Thanks everyone! :-)

[image]