apache/nuttx-apps #1643: Requesting justification for bad code pattern

slorquet, 2023-03-08:
Hello,
What is the exact reason for this wasteful array that could be replaced by a sprintf?

https://github.com/apache/nuttx/blob/master/libs/libc/string/lib_strsignal.c#L37

Thank you.

pkarashchenko, 2023-03-08:
I think with `sprintf` you can't make `strsignal` reentrant.

slorquet, 2023-03-10:
Is there really no other way to implement this? It looks particularly inefficient.

pkarashchenko, 2023-03-10:
Not that I'm aware of. You can see that other cases in switch/case also have a common `SIG` part of the string. The only difference is that in the array it is a bit longer.
Let me think a bit of what can be an option. Maybe having a reentrant variant can be a config option or we can return just the number string and drop `Signal ` prefix

pkarashchenko, 2023-06-18:
@slorquet can we close this issue now?

slorquet, 2023-06-19:
we can :)

closed 2023-06-19 by slorquet