apache/nuttx-apps #1667: NSH Login Now Exposes Password

patacongo, 2023-03-19:
Using stock configuration nsh:sim under ChromeOS (Debian), the login looks like:

    login: admin
    password: Administrator

That is an error.  The password should not be exposed.  Prior to recent breaking changes to the serial driver, the login correctly looked like:

    login: admin
    password: 

The password was not echoed so that it did not remain in the terminal display for anyone to see.

I suppose that the correct solution with the changes to the serial driver would be to disable ECHO via termios before asking for the password, and re-enabling ECHO after the password has been entering.

If TERMIOS is not enabled, then you are just screwed.  The password will always be echoed on the console.

patacongo, 2023-03-19:
It seems to me that one good solution would be behave just like Linux:  Do not special-case the console but, instead, require that the user application always configure the TTY as needed for correct operation.  This is definitely more consistent with the way that Unix is supposed to handle TTYs.

This would solve another issue:  Occasional a device other than the serial console is used for a console.  For example, a different serial port that is not the one designated as the console.  If the application configured the TTY for its particular use, then this dependence on a magic, non-standard console device would be eliminated.

xiaoxiang781216, 2023-03-19:
With https://github.com/apache/nuttx/pull/8843, the termios unrelated to hardware setting is always enabled, so the terminal aware application could always switch to/from raw mode as needed.

xiaoxiang781216, 2023-03-19:
Please try: https://github.com/apache/nuttx-apps/pull/1668 and https://github.com/apache/nuttx/pull/8843 which should fix the issue reported here.

closed 2023-03-19 by patacongo