apache/nuttx-apps #1676: "aesctr" test failing on ESP32-C3

lucasssvaz, 2023-03-21:
While testing the `esp32c3-devkit:crypto` configuration I found out that the `aesctr` is the only test that is failing. 
This issue was also confirmed by @acassis. Does anyone have any idea of why this is happening?
Thanks in advance!

Log:
```
nsh> aescbc
aescbc test ok
nsh> aesxts
OK encrypt test vector 0
OK decrypt test vector 0
OK encrypt test vector 1
OK decrypt test vector 1
OK encrypt test vector 2
OK decrypt test vector 2
OK encrypt test vector 3
OK decrypt test vector 3
OK encrypt test vector 4
OK decrypt test vector 4
OK encrypt test vector 5
OK decrypt test vector 5
OK encrypt test vector 6
OK decrypt test vector 6
OK encrypt test vector 7
OK decrypt test vector 7
OK encrypt test vector 8
OK decrypt test vector 8
OK encrypt test vector 9
OK decrypt test vector 9
OK encrypt test vector 10
OK decrypt test vector 10
OK encrypt test vector 11
OK decrypt test vector 11
OK encrypt test vector 12
OK decrypt test vector 12
OK encrypt test vector 13
OK decrypt test vector 13
nsh> des3cbc
ok, encrypt with /dev/crypto, decrypt with /dev/crypto
nsh> hmac
hmac md5 success
hmac md5 success
hmac md5 success
hmac sha1 success
hmac sha1 success
hmac sha1 success
hmac sha256 success
hmac sha256 success
hmac sha256 success
nsh> aesctr
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCGSESSION: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCGSESSION: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCGSESSION: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
10: CIOCCRYPT: Invalid argument
10: crypt with /dev/crypto failed
```

xiaoxiang781216, 2023-03-21:
@anjiahao1 please take a look.

casaroli, 2026-11-08:
Possibly fixed by apache/nuttx#18544.

closed 2026-11-08 by casaroli