apache/nuttx-apps #1812: MacOS CI failing during homebrew openssl installation.

lucasssvaz, 2023-06-28:
The MacOS CI is failing while installing openssl.

```
==> Downloading https://ghcr.io/v2/homebrew/core/u-boot-tools/blobs/sha256:d9441454506ac196e7ca0b86867cdd2b28e64efc845f37674448382d01548dba
Already downloaded: /Users/runner/work/nuttx-apps/nuttx-apps/sources/tools/homebrew/downloads/5c0c7a43db5c54cb3ba4753104693568884c41d9d856084fb2777935c8a49b5e--u-boot-tools--2023.04.monterey.bottle.tar.gz
==> Installing dependencies for u-boot-tools: openssl@3
==> Installing u-boot-tools dependency: openssl@3
==> Pouring openssl@3--3.1.1_1.monterey.bottle.tar.gz
Error: same file: /usr/local/bin/openssl and /Users/runner/work/nuttx-apps/nuttx-apps/sources/tools/homebrew/Backup/bin/openssl
Do not report this issue until you've run `brew update` and tried again.
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:1557:in `block in fu_each_src_dest'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:1574:in `fu_each_src_dest0'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:1556:in `fu_each_src_dest'
/System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/fileutils.rb:520:in `mv'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:993:in `rescue in link'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:986:in `link'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:782:in `finish'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:743:in `install_dependency'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:662:in `block in install_dependencies'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:662:in `each'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:662:in `install_dependencies'
/usr/local/Homebrew/Library/Homebrew/formula_installer.rb:408:in `install'
/usr/local/Homebrew/Library/Homebrew/upgrade.rb:203:in `install_formula'
/usr/local/Homebrew/Library/Homebrew/install.rb:358:in `install_formula'
/usr/local/Homebrew/Library/Homebrew/install.rb:303:in `block in install_formulae'
/usr/local/Homebrew/Library/Homebrew/install.rb:302:in `each'
/usr/local/Homebrew/Library/Homebrew/install.rb:302:in `install_formulae'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:274:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:94:in `<main>'
Error: Process completed with exit code 1.
```

This is happening in multiple PRs since yesterday.

acassis, 2023-06-28:
Maybe we should have a backup mirror to get those tools case they disappear from original link!

xiaoxiang781216, 2023-06-28:
Yes, it's better to achieve the essential ci tools to the reliable place.

pkarashchenko, 2023-07-02:
Possible solutions in either https://github.com/apache/nuttx/pull/9669 or https://github.com/apache/nuttx/pull/9672

btashton, 2023-07-03:
> Maybe we should have a backup mirror to get those tools case they disappear from original link!

The issue here does not have to do with upstream disappearing.

closed 2023-07-04 by pkarashchenko