apache/nuttx-apps #1989: ostest RR scheduler testing

TaiJuWu, 2023-08-22:
I have a question about RR scheduler

https://github.com/apache/nuttx-apps/blob/d73e6412b7d8453b25c91f324c81d88f8abbcd51/testing/ostest/roundrobin.c#L227
The comment shows thread_1 and thread_2 should start and complete at about the same time.
But I did not see any assert condition about it.

Is it needed to modify or it is correct?
I try to add some assert to this problem at https://github.com/TaiJuWu/nuttx-apps/tree/RR_Test
Im not sure it's useful and is this method great?
If anyone can give me some suggestion, I will be very grateful.

acassis, 2023-10-18:
@TaiJuWu I think this could be a good idea! We need more people testing RR SCHED. Did you have chance to test RR with Tickeless mode?

TaiJuWu, 2023-10-18:
No. I just used nsh:smp as my default setting. I will try using tickless mode this weekend.

TaiJuWu, 2023-10-20:
> @TaiJuWu I think this could be a good idea! We need more people testing RR SCHED. Did you have chance to test RR with Tickeless mode?

It also can work on qemu-armv8a:nsh_smp_tickless but why we need to use tickless mode?
Actually, I can't image what condition will make this test fail....

casaroli, 2026-11-08:
Fixed by #3864.

closed 2026-11-08 by casaroli