apache/nuttx-apps #2045: Enabling MbedTLS results in warnings: "MBEDTLS_SSL_DTLS_CONNECTION_ID_COMPAT" is not defined

AndreHeinemans-NXP, 2023-09-08:
When enabling CRYPTO_MBEDTLS these warnings start to appear in the build:

```In file included from mbedtls/library/ssl_misc.h:27,
                 from mbedtls/library/constant_time_internal.h:30,
                 from mbedtls/library/base64.c:25:
Error: /github/workspace/sources/apps/crypto/mbedtls/mbedtls/include/mbedtls/ssl.h:590:5: error: "MBEDTLS_SSL_DTLS_CONNECTION_ID_COMPAT" is not defined, evaluates to 0 [-Werror=undef]
  590 | #if MBEDTLS_SSL_DTLS_CONNECTION_ID_COMPAT == 0
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Using this feature in a default config is currently not possible because it will break the ci like it does in apache/nuttx#10249

One workaround is to also enable both MBEDTLS_SSL_PROTO_DTLS and MBEDTLS_SSL_DTLS_CONNECTION_ID because then MBEDTLS_SSL_DTLS_CONNECTION_ID_COMPAT gets assigned by 0 in ssl.h

closed 2023-09-09 by xiaoxiang781216

AndreHeinemans-NXP, 2023-09-13:
The problem still exists. In apache/nuttx#10249 I just enabled MBEDTLS_SSL_PROTO_DTLS and MBEDTLS_SSL_DTLS_CONNECTION_ID so the problem didn't show up anymore. These features are not used by that target and it's only there to get rid of the warnings. (Or as errors in the CI)

The issue can be easily reproduced on any target by just enabling MbedTLS and make sure that MBEDTLS_SSL_DTLS_CONNECTION_ID is not enabled.

AndreHeinemans-NXP, 2023-09-13:
Recently it even got worse. With the same settings as described above my build ended with this error:
```
arm-none-eabi-ar: flock: No such file or directory
make[2]: *** [/home/andre/projects/nuttx/apps/Application.mk:213: .built] Error 1
make[1]: *** [Makefile:51: /home/andre/projects/nuttx/apps/crypto/mbedtls_all] Error 2
make: *** [tools/LibTargets.mk:232: /home/andre/projects/nuttx/apps/libapps.a] Error 2
```

xiaoxiang781216, 2023-09-13:
flock is fixed here: https://github.com/apache/nuttx-apps/pull/2061