apache/nuttx-apps #2046: tools/Wasm.mk problems

yamt, 2023-09-08:
* the idea to build wasm modules using nuttx headers is broken
* it produces very WAMR specific modules (it seems targeting WAMR's LIBC_BUILTIN). if we want to build wasm modules, it's better to target wasm32-wasi.
* WCC detection is broken. for example, github ubuntu-22.04 runners have /bin/clang, which confuses the detection.
* CONFIG_INTERPRETERS_WASM vs CONFIG_INTERPRETERS_WASM3 typo

my proposal: remove it

an alternative:
* target it to wasm32-wasi
* have an explicit config to enable it (disable it by default)

anchao, 2023-09-08:
@yamt 
We are not using the implementation of wasi/wasi-libc. We have a python script to parse nuttx csv files([syscall](https://github.com/apache/nuttx/blob/master/syscall/syscall.csv)/[libc](https://github.com/apache/nuttx/blob/master/libs/libc/libc.csv)/[libm](https://github.com/apache/nuttx/blob/master/libs/libm/libm.csv)) to automatically generate nuttx syscall/libc/libm glue codes. Currently, this feature works as expected without boundary checking in wamr, but it still needs some work if enable boundary check.

yamt, 2023-09-11:
> @yamt We are not using the implementation of wasi/wasi-libc. We have a python script to parse nuttx csv [...]

where can i find the python script?

anchao, 2023-09-11:
> where can i find the python script?

I will upload a draft ASAP

yamt, 2023-09-11:
why do you prefer the "no sandbox" approach?
doesn't it spoil many of benefits of wasm?

no1wudi, 2023-09-12:
@yamt Please refer to this PR https://github.com/apache/nuttx-apps/pull/2058

closed 2024-10-07 by xiaoxiang781216