apache/nuttx-apps #2189: `libtest` does not work

casaroli, 2023-11-13:
Hello,

I am trying to enable the `libtest` example with config `sim:nsh` but it seems to fail to link:

```
/usr/bin/ld: nuttx.rel:(.data.rel.ro.g_builtins+0x58): undefined reference to `libtest_main'
```

It looks like `EXTRA_LIBS` is not being honored in `Make.defs`.

Does anyone know how to fix?

xiaoxiang781216, 2023-11-13:
Which Make.defs do you modify?

casaroli, 2023-11-13:
> Which Make.defs do you modify?

Hello,

I did not modify any `Make.defs`. I expect that libtest would work with the default [Make.defs](https://github.com/apache/nuttx-apps/blob/master/examples/libtest/Make.defs#L22) in `libtest`.

If I add:
```
EXTRA_LIBS = "$(APPDIR)/examples/libtest/libtest.a"
```
to `boards/sim/sim/sim/scripts/Make.defs`, then it works, but I expect it to work without changing any board files since it is already set in the example `libtest` `Make.defs`.

WDYT?

casaroli, 2023-11-14:
So the `libtest` example shipped with `nuttx-apps` does not work out of the box?

xiaoxiang781216, 2023-11-14:
libtest demonstrate that you can generate a separated library from libapps.a, but the addtional work need be done manually to link with the final nuttx image since the build system limitation.
It will be great if you can fix the problem and provide a patch.

casaroli, 2023-11-14:
Do you think it will make sense to have nuttx makefile process and include apps/Make.defs and not the other way around?

xiaoxiang781216, 2023-11-14:
Yes, it's very useful that app's Makefile could adjust EXTRA_LIB used by nuttx in some way.

casaroli, 2023-11-16:
> Yes, it's very useful that app's Makefile could adjust EXTRA_LIB used by nuttx in some way. We have tried many approach, but can't find an elegant solution.

Thank you for the reply. Please take a look at this proposal: https://github.com/apache/nuttx/pull/11202