apache/nuttx-apps #2219: esp32s3: support lowpower mode?

dapeng0512, 2023-12-06:
Hi,  I notice that the esp32s3 does not support power management. Do you have a plan for this function?

tmedicci, 2023-12-12:
Hi @dapeng0512, this feature is in our backlog but we have no due date for it to be delivered. If you can contribute to this implementation, we'll be glad to help you throughout the process ;)

FelipeMdeO, 2024-07-31:
@tmedicci  I  think we have power manager running in esp32s3 now, right? If yes, we can close this issue.

tmedicci, 2024-07-31:
Hi @dapeng0512 , could you please test https://github.com/apache/nuttx/blob/master/boards/xtensa/esp32s3/esp32s3-devkit/configs/pm/defconfig ?

Reference: https://nuttx.apache.org/docs/latest/platforms/xtensa/esp32s3/boards/esp32s3-devkit/index.html#pm

closed 2024-10-07 by xiaoxiang781216