apache/nuttx-apps #2246: [help] confirming ostest output in KERNEL mode

yf13, 2023-12-30:
Thanks for making `ostest` loadable in KERNEL mode NuttX!

I run it with kernel build on CanMV230 and got the following dumps at the end:

```
Final memory usage:                                                             
VARIABLE  BEFORE   AFTER                                                        
======== ======== ========                                                      
arena       80ff8    80ff8                                                      
ordblks         2        4                                                      
mxordblk    7cff0    787f0                                                      
uordblks     2658     3598                                                      
fordblks    7e9a0    7da60                                                      
user_main: Exiting                                                              
ostest_main: Exiting with status -1 
```

Is this expected? I understand that test cases might be different from FLAT build, but not sure on the exit code.

xiaoxiang781216, 2023-12-30:
@fxysunshine could you take a look?

yf13, 2024-01-06:
On `rv-virt/knsh64`. the exit code is also -1:

```
sched_lock: Finished

End of test memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena       80ff8    80ff8
ordblks        10        4
mxordblk    6c7f0    787f0
uordblks     65f8     3598
fordblks    7aa00    7da60

user_main: vfork() test DISABLED (CONFIG_BUILD_KERNEL)

Final memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena       80ff8    80ff8
ordblks         2        4
mxordblk    7cff0    787f0
uordblks     2658     3598
fordblks    7e9a0    7da60
user_main: Exiting
ostest_main: Exiting with status -1
nsh>
```