apache/nuttx-apps #2277: rpporxy does not work

Jan9401, 2024-01-31:


xiaoxiang781216, 2024-01-31:
could you show more log?

Jan9401, 2024-01-31:
The terminal does not have any logs. It is relatively easy to reproduce this problem.

./tool/configure.sh -l sim:rppoxy
make -j16
./nuttx

xiaoxiang781216, 2024-01-31:
you need run sim:rpserver too, please read:
https://github.com/apache/nuttx/blob/master/Documentation/platforms/sim/sim/boards/sim/index.rst#rpproxy-and-rpserver

Jan9401, 2024-01-31:
I run sim:rpserver too, But it will hang too. 
[image]
[image]

xiaoxiang781216, 2024-01-31:
@CV-Bowen could you look at this problem?

CV-Bowen, 2024-01-31:
> @CV-Bowen could you look at this problem?

Yes, i will take look today.

CV-Bowen, 2024-02-01:
@Jan9401 Hi, we make /dev/ttyproxy be /dev/console in rpproxy by using uart rpmsg, so the terminal runs ./rpproxy hang at startup is normal.
You can use `cu` command to enter the rpproxy nsh, like this:
[image]

Jan9401, 2024-02-01:
> @Jan9401 Hi, we make /dev/ttyproxy be /dev/console in rpproxy by using uart rpmsg, so the terminal runs ./rpproxy hang at startup is normal. You can use `cu` command to enter the rpproxy nsh, like [...]

OK Thanks.

closed 2024-02-02 by xiaoxiang781216