apache/nuttx-apps #2336: Some errors is reported when the sqlite database is compiled

laoniaokkk, 2024-03-22:
Hello,when I get the latest code(apps and nuttx) from the repository and then add sqlite after selecting the sim/nsh configuration, it will produce a lot of errors during compilation, such as assert, unknown type name 'off64_t', etc.
My computer is ubuntu20.04.I hope to get help to solve these problems

acassis, 2024-05-06:
@laoniaokkk please share your .config

laoniaokkk, 2024-05-07:
[config.txt](https://github.com/apache/nuttx-apps/files/15230878/config.txt)

@acassis  Hello,this is my .config, I just use 'sim/nsh' and select database as:
[*] SQLITE library                      
(3.45.1) SQLITE version                 
[*]   SQLite cmd line tool              
(8192)  SQLite3 cmd line tool stack size

acassis, 2024-05-07:
Thank you @laoniaokkk ! Using your config I got the same issue. Seems like HOST_ARM64 is not supported yet.

Then I decided to change it to HOST_X86_64 and got some errors and fixed them.

However the SQLite didn't work when I run "nsh> sqlite3". Please find the files I modified to get it compiling.

@yinshengkai @xiaoxiang781216 I appreciate if you guys could include an example using the SIM or better yet some real board.

[config_sim_sqlite.txt](https://github.com/apache/nuttx-apps/files/15238265/config_sim_sqlite.txt)
[Makefile.txt](https://github.com/apache/nuttx-apps/files/15238268/Makefile.txt)
[sqlite3.c.txt](https://github.com/apache/nuttx-apps/files/15238280/sqlite3.c.txt)

xiaoxiang781216, 2024-05-08:
@Gary-Hobson will provide patch soon.

Gary-Hobson, 2024-05-08:
Compilation error problems can be solved by adding this patch: https://github.com/apache/nuttx/pull/12303

The following patch adds an example of sqlite in sim: https://github.com/apache/nuttx/pull/12305

Since hostfs does not support FIOC_FILEPATH, it cannot currently be used in hostfs

Steps for usage:
```
./tools/configure.sh sim:sqlite
make -j 
./nuttx

nsh> cd tmp
nsh> ls
/tmp:

nsh> sqlite3 test.db
SQLite version 3.45.1 2024-01-30 16:01:20
Enter ".help" for usage hints.
sqlite> 

 CREATE TABLE COMPANY(
   ID INT PRIMARY KEY     sqlite> (x1...> NOT NULL,
   NAME           TEXT    NOT NULL,
   AGE            (x1...> (x1...> INT     NOT NULL,
   ADDRESS        CHAR(50),
   SALARY         (x1...> (x1...> REAL
);(x1...> 

sqlite> .schema COMPANY
CREATE TABLE COMPANY(
   ID INT PRIMARY KEY     NOT NULL,
   NAME           TEXT    NOT NULL,
   AGE            INT     NOT NULL,
   ADDRESS        CHAR(50),
   SALARY         REAL
);
sqlite> .quit
sqlite> 
nsh> 
nsh> ls -l
/tmp:
 -rwxrwxrwx       12288 test.db
```

casaroli, 2026-11-08:
Possibly fixed by apache/nuttx#12303 and apache/nuttx#12305.

closed 2026-11-08 by acassis