apache/nuttx-apps #2461: LVGL Version

TimJTi, 2024-08-05:
# Summary

NuttX Apps needs to be upgraded to the newly released V9.2 to solve framebuffer render issues.

## Background

As discussed [here](https://github.com/apache/nuttx-apps/issues/2416) the LVGL version "bundled" with NuttX apps (v9.1) has issues with what I perceive as its misuse of the NuttX framebuffer; it makes multiple partial renders **_direct_** to the framebuffer, resulting in visible artefacts. It is deemed OK as there are some architectures/boards where this is OK because it seems they natively have two framebuffers. But - to me - it is the exception not the rule. 

V9.2 now auto-detects the presence of that second native framebuffer and if its not there creates its own render buffer. It works but is perhaps a bit arcane.

NB - there is no support for an LVGL partial render buffer in the LVGL NuttX framebuffer driver, so there will be two whole frame buffers present: the /dev/fb0 buffer and LVGLs render buffer.

### Other

While investigating this I needed to work with other versions of LVGL and found the following.

1. NuttX "fixes" the LVGL version.
2. If you clone LVGL direct to the appropriate NuttX apps directory you can use any version you like. This could perhaps be added to documentation, or:
  1a. Enable Kconfig configuration of the wanted LVGL version. Not sure this works if you need to work with LVGL master as there will be no release archive that can be downloaded.
3.  **BUT** the side effect of this that NuttX apps takes a direct copy of LVGLs Kconfig from the "fixed"/"approved" version of a given NuttX apps release and this requires a manual copy if the non-approved version's Kconfig file. This has been discussed before but surely there's a way to work around this - perhaps a warning if the version is edited to something other than the "supported" version?

Discuss :-)

xiaoxiang781216, 2024-09-19:
is lvgl 9.2 released?

cederom, 2024-09-19:
> is lvgl 9.2 released?

Yes official release is ready :-) https://github.com/lvgl/lvgl/releases/tag/v9.2.0

XuNeo, 2024-09-20:
>  perhaps a warning if the version is edited to something other than the "supported" version?

Will add in https://github.com/lvgl/lvgl/pull/6789

cederom, 2024-10-06:
Okay lets wait for LVGL 9.2.1 release, verify it, and add to NuttX 12.7.0 release, so far testing on 12.7.0-RC0 failed see https://github.com/apache/nuttx/issues/13825 and https://github.com/apache/nuttx/issues/13839.

cederom, 2024-10-18:
Okay the LVGL `release/v9.2` branch is here and this will contain 9.2.1 release code that is 99% ready we can proceed :-)

https://github.com/lvgl/lvgl/tree/release/v9.2

closed 2024-10-25 by jerpelea