apache/nuttx-apps #2505: [BUG] wapi_set_ifdown failed

metarutaiga, 2024-08-19:
### Description / Steps to reproduce the issue

https://github.com/apache/nuttx-apps/blob/9bfe138087ebc1d2439aeb252b6b26e347ad9759/wireless/wapi/src/network.c#L250

It miss to set IFF_DOWN into ifr.ifr_flags;
```
       ifr.ifr_flags &= ~IFF_UP;
+      ifr.ifr_flags |= IFF_DOWN;
```

### On which OS does this issue occur?

[Mac]

### What is the version of your OS?

macOS 14

### NuttX Version

12.6.0

### Issue Architecture

[risc-v]

### Issue Area

[Networking]

### Verification

- [X] I have verified before submitting the report.

acassis, 2024-09-19:
@tmedicci what do you think? Is this modification correct?

xiaoxiang781216, 2024-10-07:
@acassis @metarutaiga the problem is fixed by https://github.com/apache/nuttx-apps/pull/2647 and https://github.com/apache/nuttx/pull/13842.

closed 2024-10-07 by GUIDINGLI