apache/nuttx-apps #2568: [BUG] Wrong `lcp_state` flag handling prevents PPP from connecting to peer after peer disconnection/re-connection

lzungri, 2024-09-13:
### Description / Steps to reproduce the issue

If a PPP peer disconnects and then tries to reconnect it will send an `'LCP configure request'` packet. The code that handles that scenario seems to be clearing the wrong `lcp_state` flag (`LCP_RX_UP` instead of `LCP_TX_UP`) and thus the nuttx ppp client will keep sending IPCP packets which are rightfully dropped by the new peer since it is still in the LCP negotiation phase.

Consider replacing this code:
```
ctx->lcp_state &= ~LCP_RX_UP;
```

with:
```
ctx->lcp_state &= ~LCP_TX_UP;
```

Source here: https://github.com/apache/nuttx-apps/blob/1f51bfe64dfa931d97137b0c7561c053e1d6078d/netutils/pppd/lcp.c#L138

I can create a PR if that helps.

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Ubuntu

### NuttX Version

12.6

### Issue Architecture

[all]

### Issue Area

[Networking]

### Verification

- [X] I have verified before submitting the report.

xiaoxiang781216, 2024-09-13:
@lzungri could you provide a patch?

lzungri, 2024-09-13:
PR here #2571

closed 2024-09-14 by xiaoxiang781216