apache/nuttx-apps #2632: [BUG] Implicit definiton of accept4 function when building simulator on Arch Linux

linguini1, 2024-09-29:
### Description / Steps to reproduce the issue

I was attempting to build the Simulator with the `tcpblaster` configuration to test out some NuttX networking, but I ran
into an implicit function definition error:

```console
[linguini@pastabox nuttx]$ make -j
CC:  tcpblaster_server.c
LN: platform/board to /home/linguini/cuinspace/pico-nuttx/apps/platform/dummy
tcpblaster_server.c: In function ‘tcpblaster_server’:
tcpblaster_server.c:159:14: error: implicit declaration of function ‘accept4’; did you mean ‘accept’? [-Wimplicit-function-declaration]
  159 |   acceptsd = accept4(listensd, (FAR struct sockaddr *)&myaddr, &addrlen,
      |              ^~~~~~~
      |              accept
make[3]: *** [Makefile:92: tcpblaster_server.hobj] Error 1
make[2]: *** [Makefile:53: /home/linguini/cuinspace/pico-nuttx/apps/examples/tcpblaster_context] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:180: context] Error 2
make: *** [tools/Unix.mk:457: /home/linguini/cuinspace/pico-nuttx/apps/.context] Error 2
make: *** Waiting for unfinished jobs....
```

Looking at the manpages for `accept4`, it seems to require that `_GNU_SOURCE` be defined before the header is included.
I added `-D_GNU_SOURCE` to this line of the TCP blaster example, and it fixed the build issue on my machine. Others that
I am working with were able to build the example without any issues on Ubuntu, so I wanted to check and see if I am
perhaps taking the wrong approach before I open a PR to permanently add that change to the example.

I followed the first three steps [listed in the simulator docs here](https://nuttx.apache.org/docs/latest/guides/simulator.html#accessing-the-network).
```console
make distclean
./tools/configure.sh sim:tcpblaster
make
```

### On which OS does this issue occur?

[Linux]

### What is the version of your OS?

Linux 6.10.10-arch1-1 x86_64 GNU/Linux

### NuttX Version

master

### Issue Architecture

[simulator]

### Issue Area

[Applications], [Build System], [Configuring], [Networking]

### Verification

- [X] I have verified before submitting the report.

cederom, 2024-09-29:
Thanks @linguini1 yes this problem was introduced recently by https://github.com/apache/nuttx-apps/pull/2595 we are looking for a solution should show up soon :-)

linguini1, 2024-09-29:
> Thanks @linguini1 yes this problem was introduced recently by https://github.com/apache/nuttx-apps/pull/2595 we are looking for a solution should show up soon :-)

Ah, I missed seeing that! Would it be preferable for me to open a PR to add this new define, or shall I leave the solution to the original author?

cederom, 2024-09-29:
Lets wait for the original change author to reply.. `accept4` is a non-standard extension that increases security.. we already noticed problems on macOS that does not have that call implemented.. no clue how `_GNU_SOURCE` will impact other compilers / compatibility avalanche.. but your hint may be right the solution!! :-)

xiaoxiang781216, 2024-09-30:
we can call accept4 on nuttx, but call accept on host.

closed 2024-10-05 by xiaoxiang781216