apache/nuttx-apps #2812: [CI] checkpatch fails.

cederom, 2024-11-01:
### Description / Steps to reproduce the issue

Our CI checkpatch fails.. @lupyuen could you please take a look?
```
Run echo "::add-matcher::nuttx/.github/nxstyle.json"
Collecting cmake-format
  Downloading cmake_format-0.6.13-py3-none-any.whl (19 kB)
Collecting cmakelang>=0.6.13
  Downloading cmakelang-0.6.13-py3-none-any.whl (159 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 159.8/159.8 KB 13.1 MB/s eta 0:00:00
Collecting six>=1.13.0
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Installing collected packages: six, cmakelang, cmake-format
Successfully installed cmake-format-0.6.13 cmakelang-0.6.13 six-1.16.0
ccd3ee983 Merge d2a2621f34c945ee52a4d498f77a4b1650cc48b4 into 242b9473423cd29aa4be6fe8efd4c4f33317bcdd
d2a2621f3 Improve nxdiag example for Espressif devices
../nuttx/tools/checkpatch.sh -m -g 242b9473423cd29aa4be6fe8efd4c4f33317bcdd..HEAD
../nuttx/tools/checkpatch.sh: line 87: black: command not found
../nuttx/tools/checkpatch.sh: line 88: flake8: command not found
../nuttx/tools/checkpatch.sh: line 89: isort: command not found
../nuttx/tools/checkpatch.sh: line 92: isort: command not found
Error: Process completed with exit code 1.
```

* If we use Debian/Ubuntu on the build machine I recently noticed they chanhed the `/bin/sh` link from `/bin/bash` to `/usr/bin/dash` and DASH is not quite compatible with BASH lol :-) It is safest to use `/usr/bin/env bash` now ;-)
* Maybe missing packages needs to be installed?

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

GitHub CI Ubuntu?

### NuttX Version

master

### Issue Architecture

[Arch: all]

### Issue Area

[Area: Tools]

### Verification

- [X] I have verified before submitting the report.

lupyuen, 2024-11-02:
Sorry @simbit18 do you recall how checkpatch works? I remember seeing this error in the NuttX Repo, I fear  that we missed something when syncing the CI Workflow from NuttX Repo to NuttX Apps. Thanks!

lupyuen, 2024-11-02:
I think we need to sync this from NuttX Repo to NuttX Apps:
- https://github.com/apache/nuttx/pull/14323

cederom, 2024-11-02:
Thank you @lupyuen !! :-)

closed 2024-11-02 by cederom