apache/nuttx-apps #2836: [BUG] User led Rust example doesn't compile

lvanasse, 2024-11-10:
### Description / Steps to reproduce the issue

To reproduce the issue:
1. Configure for the `nucleo-l452re` config: ./tools/configure.sh -E -l nucleo-l452re:nsh
2. In the `make menuconfig` enable the `USERLED` option for `LED Driver`
3. In the `make menuconfig` enable the `EXAMPLES_LEDS_RUST` option, and leave everything as default
4. Compile with `make -j`

And you should have this error message:

```bash
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/nuttx/staging/libapps.a(leds_rust_main.rs.home.ludovic.Code.nuttx_ws.apps.examples.leds_rust_1.o): in function `leds_rust_main::nuttx::safe_puts':
/home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:108: undefined reference to `__aeabi_memclr'
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:108: undefined reference to `__aeabi_memclr'
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/nuttx/staging/libapps.a(leds_rust_main.rs.home.ludovic.Code.nuttx_ws.apps.examples.leds_rust_1.o): in function `leds_rust_main::nuttx::safe_open':
/home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:77: undefined reference to `__aeabi_memclr'
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/nuttx/staging/libapps.a(leds_rust_main.rs.home.ludovic.Code.nuttx_ws.apps.examples.leds_rust_1.o): in function `leds_rust_main::nuttx::safe_puts':
/home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:108: undefined reference to `__aeabi_memclr'
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:108: undefined reference to `__aeabi_memclr4'
arm-none-eabi-ld: /home/ludovic/Code/nuttx_ws/apps/examples/leds_rust/nuttx.rs:108: undefined reference to `__aeabi_memclr'
make[1]: *** [Makefile:212: nuttx] Error 1
make: *** [tools/Unix.mk:551: nuttx] Error 2
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 22.04 LTS

### NuttX Version

master - e2a21337ac5be663a8e6d9a82baa64b93cc19755 and stable 12.7

### Issue Architecture

[Arch: arm]

### Issue Area

[Area: Examples]

### Verification

- [X] I have verified before submitting the report.

lvanasse, 2024-11-10:
I've asked on the Discord server and as always @lupyuen was very useful and giving me ideas to test :), here's the transcript of his response:

> (1) Isn't __aeabi_memclr defined in glibc? https://codebrowser.dev/glibc/glibc/sysdeps/arm/aeabi_memclr.c.html
(2) Could you run make --trace to see which libraries we're linking with?
(3) Can you reproduce this problem with qemu-armv7a:nsh? https://nuttx.apache.org/docs/latest/platforms/arm/qemu/boards/qemu-armv7a/index.html
(4) no1wudi is doing great work on the upcoming Rust Standard Library for NuttX, which is way more advanced than the Rust Core Library that's called by leds_rust. As discussed here: https://lupyuen.github.io/articles/rust6#all-things-considered 
(5) Rust Core Library is a little wonky on NuttX, but it ought to work. Let's track down this bug thanks 🙂

lvanasse, 2024-11-10:
Here's the `make -j -trace` output:

https://gist.github.com/lvanasse/f62e9fa59f8382dd9881178fc1904211

lvanasse, 2024-11-10:
When trying for the `qemu-armv7a:nsh`

```bash
nuttx_ws/nuttx [master●] » make -j 
error[E0463]: can't find crate for `core`
  |
  = note: the `armv7a-none-eabihf` target may not be installed
  = help: consider downloading the target with `rustup target add armv7a-none-eabihf`

error: aborting due to 1 previous error

For more information about this error, try `rustc --explain E0463`.
make[2]: *** [/home/ludovic/Code/nuttx_ws/apps/Application.mk:330: leds_rust_main.rs.home.ludovic.Code.nuttx_ws.apps.examples.leds_rust.o] Error 1
make[1]: *** [Makefile:52: /home/ludovic/Code/nuttx_ws/apps/examples/leds_rust_all] Error 2
make: *** [tools/LibTargets.mk:248: /home/ludovic/Code/nuttx_ws/apps/libapps.a] Error 2
make: *** Waiting for unfinished jobs....
nuttx_ws/nuttx [master●] » 
```

And when I try to download the target here's the message I get:

```bash
nuttx_ws/nuttx [master●] » rustup target add armv7a-none-eabihf
error: toolchain 'stable-x86_64-unknown-linux-gnu' does not support target 'armv7a-none-eabihf'; did you mean 'armv7r-none-eabihf'?
note: you can see a list of supported targets with `rustc --print=target-list`
note: if you are adding support for a new target to rustc itself, see https://rustc-dev-guide.rust-lang.org/building/new-target.html
```

lupyuen, 2024-11-10:
@lvanasse I was able to reproduce the problem with Docker: https://gist.github.com/nuttxpr/a881eea7923c9471d026ac2bfcbdac82

It seems the problem is caused by the incompatibility between Rust Compiler and GCC Linker used by NuttX:
- Rust Compiler emits LLVM Symbols like __aeabi_memclr, expecting them to be in libc (C Standard Library) from the LLVM Linker
- But for GCC Linker on Embedded Platforms: These LLVM Symbols [don't exist](https://gist.github.com/nuttxpr/a881eea7923c9471d026ac2bfcbdac82#file-gistfile1-txt-L25368)
  ```bash
  ## __aeabi_memclr doesn't exist in libgcc
  $ objdump -t  /tools/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/thumb/v7e-m/nofp/libgcc.a  \
    | grep memclr
  ```
- Some Embedded Folks wrote their own implementation of __aeabi_memclr: [Rust for STM32](https://rudi-horn.de/Home/6-rust-stm-handling-static-variables) / [Rust for TockOS](https://www.hackster.io/alevy/building-your-first-tock-app-540695#:~:text=There%20are%20known%20issues%20with%20arm%2Dnone%2Deabi%2Dgcc%20version%205.1%20and%20older%2C%20or%20other%20versions%20packaged%20with%20a%20newlib%20version%20earlier%20than%202.3%2C%20as%20they%20will%20run%20into%20problems%20with%20missing%20ARM%20intrinsics%20(e.g.%2C%20__aeabi_memclr%20).%20Tock%20does%20not%20support%20these%20versions.)

So it seems we have to wait for NuttX support to become official in Rust:
- https://github.com/apache/nuttx/issues/12960

lvanasse, 2024-11-10:
You're awesome @lupyuen ! I'm a bit ashamed I was not to be able to figure this out on my own, thank you very much for helping and teaching me :).

Is there a way I could be useful in the process of making this possible to use on ARM? Could @no1wudi use some help?

Maybe we could do our own call in the Rust code for the `__aeabi_memclr` call? Like they do in the [Rust for STM32](https://rudi-horn.de/Home/6-rust-stm-handling-static-variables) article?

lupyuen, 2024-11-10:
@lvanasse No worries :-) Yep we could code it ourselves. Actually it could have been simpler if we used Rust Libraries (Crates), but NuttX doesn't support them sigh: https://lupyuen.github.io/articles/rust6#no-crates-in-nuttx

The best we can do right now: Put the code into a Rust Source File (like nuttx.rs) and share it across our Rust Projects. Like this: https://github.com/apache/nuttx-apps/blob/master/examples/leds_rust/nuttx.rs

lvanasse, 2024-11-10:
Awesome! Thank you for the guidance, I'll try to give it a go when I have the energy/time for it :).