apache/nuttx-apps #2884: [FEATURE] Python on NuttX: "To-Do" list and general enhancements

tmedicci, 2024-12-09:
### Is your feature request related to a problem? Please describe.

After https://github.com/apache/nuttx-apps/pull/2879 was merged, it's time to make Python on NuttX fully usable!

This issue intends to discuss the next steps required to make it fully functional, including a "To-Do" list and any discussions on how it would evolve to support other architectures and platforms.

## Initial Considerations

1. Python for NuttX target initially the `rv-virt` (RISC-V QEMU). That being said, its `interpreters/python/config.site` and `interpreters/python/Setup.local` are somehow tied with the features enabled by the `boards/risc-v/qemu-rv/rv-virt/configs/cpython/defconfig`. Python for NuttX depends on `CONFIG_EXPERIMENTAL`!
2. Python modules are stored in `pyc` (byte-code format) and are loaded from a ROMFS image at startup.
3. Environment variables like `PYTHONHOME` and `PYTHON_BASIC_REPL`  needs to be set accordingly.

### Describe the solution you'd like

## TO-DO list:

This is the list of the next steps to make Python's port fully functional (sorted by highest priority):

- [ ] Provide a CMake-based build: currently we can only build it using `Make`.
- [ ] Generate `interpreters/python/Setup.local` and  `interpreters/python/config.site` dynamically according to the enabled configs. For instance: [`export ac_cv_func_pipe="yes"`](https://github.com/apache/incubator-nuttx-apps/blob/efc1bf710c8be1e0f0ae5e1bf2b292f8b57076a6/interpreters/python/config.site#L20) depends on `CONFIG_PIPES` being enabled.
- [ ] Create a wrapper application to automate the 1) mounting the ROMFS image with the Python's modules, 2) setting the environment variables and 3) running the `Program/python.c` app.

## Known Issues
- [ ] Check why enabling the optimization fails to generate coherent code for `Python/Modules/getpath.c `. Reported at https://github.com/apache/incubator-nuttx-apps/blob/efc1bf710c8be1e0f0ae5e1bf2b292f8b57076a6/interpreters/python/Makefile#L86 (valid for `riscv-none-elf-gcc (xPack GNU RISC-V Embedded GCC x86_64) 13.2.0`)

### Verification

- [X] I have verified before submitting the report.

tmedicci, 2025-01-07:
Update: Python's socket module is now supported and Python on NuttX can run applications with POSIX-compatible sockets. Please check https://github.com/apache/nuttx/pull/15457 and https://github.com/apache/nuttx-apps/pull/2928 for examples.

tmedicci, 2025-01-08:
Hi, I prepared an article with the technical details of the implementation in my blog. You can check it at:

https://tmedicci.github.io/articles/2025/01/08/python_on_nuttx.html

tmedicci, 2025-01-31:
Just submitted a new PR (https://github.com/apache/nuttx-apps/pull/2982) that solves two issues on `TO-DO list`:

1. Generate interpreters/python/Setup.local and interpreters/python/config.site dynamically according to the enabled configs. For instance: [export ac_cv_func_pipe="yes"](https://github.com/apache/incubator-nuttx-apps/blob/efc1bf710c8be1e0f0ae5e1bf2b292f8b57076a6/interpreters/python/config.site#L20) depends on CONFIG_PIPES being enabled.
2. Create a wrapper application to automate the 1) mounting the ROMFS image with the Python's modules, 2) setting the environment variables and 3) running the Program/python.c app.

arnavsharma990, 2026-09-17:
Hi @tmedicci — #2884 lists "Provide a CMake-based build" unchecked... May I take the CMake piece as a focused nuttx-apps-only PR?