apache/nuttx-apps #2966: [FEATURE] GPS example autoselect MINMEA and configurable serial port

linguini1, 2025-01-22:
### Is your feature request related to a problem? Please describe.

Only /dev/ttyS1 works with the GPS example, and it doesn't auto-select its dependencies.

### Describe the solution you'd like

The GPS example should automatically select `CONFIG_GNSSUTILS_MINMEA_LIB` since it depends on it to build.

Additionally, instead of being hard-coded to read a GPS on `/dev/ttyS1`, the path to the serial port should be able to be passed as a command line argument (so we can support systems that use `/dev/ttyS0` for their GPS).

### Describe alternatives you've considered

Modifying the hard-coded value before compiling, which isn't a great user experience. Selecting the MINMEA option isn't difficult, but I would expect dependencies to be enabled when I enable the GPS example intuitively.

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-01-22:
please add `depends on GNSSUTILS_MINMEA_LIB`

xiaoxiang781216, 2025-01-22:
BTW, uORB include the special support for uORB you can try too.

PR #2969 (2025-01-23): Allow GPS serial port to be specified as argument to `gps` example
## Summary

This PR adds the ability to specify the GPS's serial port in the `gps` example, instead of only being able to use `/dev/ttyS1`. It also marks the MINMEA library as a dependency for the `gps` example.

## Impact

Now that MINMEA is marked as a dependency, the `gps` example is unfortunately not visible in the list of applications until MINMEA is available. However, I wanted to avoid the use of `select` in Kconfig since it will not recursively enable dependencies.

The user is now also able to run the `gps` example wit [... cut]
closed 2025-01-23 by xiaoxiang781216