apache/nuttx-apps #2991: [FEATURE] Should we extend fastboot oem to support executing custom commands?

JianyuWang0623, 2025-02-11:
### Is your feature request related to a problem? Please describe.

Should we extend "fastboot oem" to support executing custom commands for test or debug, such as, make ram disk: `fastboot oem sh "mkrd -m 10 -s 512 640"`. -- https://github.com/apache/nuttx-apps/blob/master/system/fastboot/fastboot.c#L200

The background/reason:
1. Enabled fastboot for NuttX based bootloader, and plan to test fastboot commands that will write data to block device. But there is no partition on devices created for fastboot testing generally, so we need making ram disk maybe.
2. The nsh was exited/disabled so we can not input command ("mkrd" for making ram disk).
3. ADB and Fastboot can not enable at the same time currently, so we can not executing command through ADB. -- https://github.com/apache/nuttx/blob/master/drivers/usbdev/adb.c#L53 

### Describe the solution you'd like

Add support for "fastboot oem sh" to support executing custom commands.

### Verification

- [x] I have verified before submitting the report.

cederom, 2025-02-11:
I like the idea :-) Should it be build-time selectable / configurable? It may introduce security issue when commands execution is possible at this point on production systems? :-)

JianyuWang0623, 2025-02-24:
@cederom Could you review this pull request (https://github.com/apache/nuttx-apps/pull/3003) please?

cederom, 2025-02-24:
> @JianyuWang0623  [@cederom](https://github.com/cederom) Could you review this pull request ([#3003](https://github.com/apache/nuttx-apps/pull/3003)) please?

done :-) thanks! :-)