apache/nuttx-apps #3058: [BUG] elf example fails to compile on esp32-devkitc

Laczen, 2025-04-15:
### Description / Steps to reproduce the issue

The elf example fails to compile for esp32-devkitc. I am unsure if this is related to the latest changes to libelf.

To reproduce:
```console
./tools/configure.sh -l esp32-devkitc:elf
...
make flash ESPTOOL_PORT=/dev/ttyUSB0 -j 8
Create version.h
Cloning Espressif HAL for 3rd Party Platforms
LN: platform/board to /home/batman/nuttxspace/apps/platform/dummy
Register: hello
Register: elf
Register: sh
Register: nsh
Espressif HAL for 3rd Party Platforms: a461ca0750d1a3deca6b10a283064dbcd2b76fb1
CC:  mqueue/mq_rcvinternal.c CC: errno.c
CC:  chip/esp32_irq.c LD: errno.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  dirent/lib_fdopendir.c CC: hello.c
LD: hello.o
CC:  dirent/lib_closedir.c xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC: struct_main.c
CC: struct_dummy.c
CC:  mm_heap/mm_foreach.c CC: signal.c
CC:  chip/esp32_libc_stubs.c LD: struct_main.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  dirent/lib_seekdir.c CC: hello++1.cxx
LD: signal.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC: hello++2.cxx
CC:  chip/esp32_idle.c CC: hello++3.cxx
LD: hello++1.o
LD: hello++2.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  chip/esp32_efuse.c LD: hello++3.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  grp/lib_getgrgidr.c CC: mutex.c
CC: pthread.c
CC:  nsh_builtin.c LD: mutex.o
LD: pthread.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
CC:  umm_heap/umm_globals.c CC: task.c
CC:  pthread/pthread_create.c LD: task.o
xtensa-esp32-elf-ld: warning: cannot find entry symbol __start; not setting start address
```

### On which OS does this issue occur?

[OS: Linux]

### What is the version of your OS?

Ubuntu 24.01

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Binfmt]

### Verification

- [x] I have verified before submitting the report.

xiaoxiang781216, 2025-04-15:
dup with https://github.com/apache/nuttx/issues/16201, let's track the problem there.

closed 2025-04-15 by xiaoxiang781216

Laczen, 2025-04-15:
Oops, sorry for duplicate issue.

xiaoxiang781216, 2025-04-15:
no problem, please try the patch whether can fix your problem and report your result there.