apache/nuttx-apps #3111: [BUG] netutils/thttpd CGI broken

TimJTi, 2025-06-26:
### Description / Steps to reproduce the issue

As best as I can tell, the CGI script handling side of THTTPD is broken.

The server can deliver webpages with no problem (as long as you guess all the Kconfig settings it needs), and correctly detects calls to CGI applications and runs them. All the data is passed to the CGI app (and I can save a file from a POST handler script/app) but if the app attempts to send data back to the calling webpage (e.g. http status codes) the actual write fails completely as it deems the stdout pipe to have incorrect write privileges.

This is true of the example app "hello" CGI app and also for a few I created. They all behave the same.

Either it is broken, or there's a magic Kconfig setting I have not fathomed out. I need fairy dust.

For now, I have given up trying to debug it as it is just behaving weirdly, but I would be interested to know if anyone is actually using it for CGI scripts successfully?

I have gone back to trying to use uIP webserver which, by design, is woefully inadequate, only surporting GET methods, but I will see if I can coerce it to respond to a PUT (which maps to GET in html 5) or POST (which is outright rejected at the moment), either by minor enhancement to it or doing what I want with a REST API. But I had really, and not unreasonably I think, hoped that NuttX would offer a working, integrated, webserver for me to use "out of the box".

```

#
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_ACT8945A_LDO1_PULLDOWN is not set
# CONFIG_ACT8945A_LDO2_BOOT_ON is not set
# CONFIG_DEV_RANDOM is not set
# CONFIG_DISABLE_OS_API is not set
# CONFIG_FS_ANONMAP is not set
# CONFIG_LV_BUILD_EXAMPLES is not set
# CONFIG_LV_FONT_MONTSERRAT_14 is not set
# CONFIG_NETUTILS_DHCPD_IGNOREBROADCAST is not set
# CONFIG_NSH_ALIAS is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
# CONFIG_NSH_DISABLE_DATE is not set
# CONFIG_SAMA5D27_JTIT2_PROTECT_BOOTFLASH is not set
# CONFIG_SAMA5_PWM_CHAN0_OUTPUTH is not set
# CONFIG_SAMA5_PWM_CHAN2_OUTPUTH is not set
# CONFIG_SAMA5_PWM_CHAN3_OUTPUTH is not set
# CONFIG_SAMA5_RXLP is not set
# CONFIG_SAMA5_UART0 is not set
# CONFIG_SIG_SIGSTOP_ACTION is not set
# CONFIG_SYSTEM_DD is not set
# CONFIG_SYSTEM_SETTINGS_CACHED_SAVES is not set
CONFIG_ACT8945A_DCDC1_NAME="VDD_IODDR"
CONFIG_ACT8945A_DCDC2_NAME="VDD_CORE"
CONFIG_ACT8945A_DCDC3_NAME="VDD_3V3"
CONFIG_ACT8945A_LDO1_NAME="VDD_FUSE"
CONFIG_ACT8945A_LDO2_NAME="VDD_3V3_LP"
CONFIG_ACT8945A_LDO3_APPLY_UV=y
CONFIG_ACT8945A_LDO3_BOOT_ON=y
CONFIG_ACT8945A_LDO3_MAX_UV=3300
CONFIG_ACT8945A_LDO3_MIN_UV=3300
CONFIG_ACT8945A_LDO3_NAME="VDD_LED"
CONFIG_ACT8945A_LDO4_NAME="VDD_GPS_ANTENNA"
CONFIG_ADC_FIFOSIZE=40
CONFIG_APDS9922_ALS_NPOLLWAITERS=4
CONFIG_APDS9922_PS_NPOLLWAITERS=4
CONFIG_ARCH="arm"
CONFIG_ARCH_BOARD_CUSTOM=y
CONFIG_ARCH_BOARD_CUSTOM_DIR="../CustomBoards/jti-toucan2"
CONFIG_ARCH_BOARD_CUSTOM_DIR_RELPATH=y
CONFIG_ARCH_BOARD_CUSTOM_NAME="jti-toucan2"
CONFIG_ARCH_CHIP="sama5"
CONFIG_ARCH_CHIP_ATSAMA5D27=y
CONFIG_ARCH_CHIP_SAMA5=y
CONFIG_ARCH_CHIP_SAMA5D2=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_LOWVECTORS=y
CONFIG_ARM_FPU_ABI_SOFT=y
[... 389 lines cut ...]
  -isystem /home/tim/designs/nuttx/nuttx/include/cxx
  -isystem /home/tim/designs/nuttx/nuttx/include
  -D__NuttX__ 
  -pipe

NuttX LDFLAGS:
  --print-memory-usage
  --no-warn-rwx-segments
  --entry=__start
  -nostdlib
  --gc-sections
  --cref
  -Map=/home/tim/designs/nuttx/nuttx/nuttx.map

NuttX configuration options:
  CONFIG_HOST_LINUX=y

Host system OS:
  CONFIG_APPS_DIR=\

Host system PATH:

### Verification

- [x] I have verified before submitting the report.

TimJTi, 2025-06-27:
I have done a little investigation and don't know if i have found out anything that might be a clue so would really appreciate some help here.

When the CGI task is executed it pipes/dups stdin and stdout. The CGI app is also passed the file descriptor of the open socket which it ultimately uses to copy the CGI app's buffered http output on to the connected browser.

When stdout is being duplicated, one of the file descriptors being allocated is the *same* as the socket file descriptor so - to me - it seems not surprising it fails to work?

Why would a socket's file descriptor be the same as a a file descriptor that dup2 hands out?

Am I missing a Kconfig setting to make this behave? Or is this a bug? Or me misunderstanding something and the problem lies elsewhere?

Any suggestions welcomed (I have tried master and 12.9)

TimJTi, 2025-06-29:
OK...the file descriptor for the socket is closed during the exec() of the CGI app allowing it to be resued by dup2() and is invalid for use by the CGI app. This is assuming the NuttX documentation is correct in saying only the first 3 file descriptors will be valid after the exec.

I will try changing the exec() to a posix_spawn() as that should allow the open socket FD to be used correctly and report back and/or submit a PR.

xiaoxiang781216, 2025-07-02:
> I have found the issue and will submit a PR shortly:
> 
> The thttpd socket needs to be opened without the SOCK_CLOEXEC flag, as a CGI script needs access to the socket after the CGI task is created.
> 
> This is the result, I believe, of a PR last year [here](https://github.com/apache/nuttx-apps/pull/2595)

@TimJTi it's wrong to add SOCK_CLOEXEC at
https://github.com/apache/nuttx-apps/pull/2595/files#diff-7b30545ee837b42152a77d494a57bae1340948ec098f6df957e344b1938920cfR2358 since the child task created at https://github.com/apache/nuttx-apps/blob/master/netutils/thttpd/thttpd_cgi.c#L1060 expect to inheritance conn_fd.

TimJTi, 2025-07-02:
See [PR](https://github.com/apache/nuttx-apps/pull/3119).