apache/nuttx-apps #3142: [BUG] mbedtls "alt" functions return errors in a wrong way

yamt, 2025-07-25:
### Description / Steps to reproduce the issue

our mbedtls alt functions (eg. https://github.com/apache/nuttx-apps/blob/72668d723447e79a73e850da1956b6edc141fd17/crypto/mbedtls/source/sha1_alt.c#L54-L63)
return negative nuttx errno, while they should return mbedtls error code like MBEDTLS_ERR_SHA1_BAD_INPUT_DATA.

### On which OS does this issue occur?

[OS: Mac]

### What is the version of your OS?

macOS 15.5

### NuttX Version

master

### Issue Architecture

[Arch: xtensa]

### Issue Area

[Area: Crypto]

### Verification

- [x] I have verified before submitting the report.

ThePassionate, 2025-07-28:
@yamt https://github.com/apache/nuttx-apps/pull/3144 fix this issue and aligned error value with mbedtls test suits
firstly, with introduce of shax_update in header file like
[https://github.com/Mbed-TLS/mbedtls/blob/v3.6.4/include/mbedtls/sha1.h#L130](url)
There is no strict rule that the return value of this function will be an error. This is because the buindin implementation will not be wrong, so error value will not be obtained. However, in the official test set, there is indeed an illegal parameter check for the judgment parameters of the 'shax_starts' function. so i add the input value check and show in following
`nsh> test_suite_shax`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/1 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/2 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/3 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/4 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/5 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/6 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/7 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/8 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/9 .................................... PASS`
`SHA-1 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/10 ................................... PASS`
`SHA-256 Invalid parameters ........................................ PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/1 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/2 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/3 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/4 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/5 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/6 .................................. PASS`
`SHA-224 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/7 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/1 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/2 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/3 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/4 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/5 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/6 .................................. PASS`
`SHA-256 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/7 .................................. PASS`
`SHA-512 Invalid parameters ........................................ PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/1 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/2 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/3 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/4 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/5 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/6 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/7 .................................. PASS`
`SHA-384 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/8 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/1 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/2 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/3 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/4 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/5 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/6 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/7 .................................. PASS`
`SHA-512 Test Vector NIST CAVS https://github.com/apache/nuttx-apps/pull/8 .................................. PASS`

yamt, 2025-07-28:
> [@yamt](https://github.com/yamt) [#3144](https://github.com/apache/nuttx-apps/pull/3144) fix this issue and aligned error value with mbedtls test suits

thank you.

> firstly, with introduce of shax_update in header file like
> [https://github.com/Mbed-TLS/mbedtls/blob/v3.6.4/include/mbedtls/sha1.h#L130](url)
> There is no strict rule for the return value of this function.

while there might not be strict rules, i guess we should avoid confusing users.
negative nuttx errno values badly conflict with MBEDTLS_ERR_xxx values and
confuses mbedtls_strerror.
for example, -EXDEV == -18 == MBEDTLS_ERR_GCM_AUTH_FAILED.

ThePassionate, 2025-07-28:
> thank you.
>
>
> while there might not be strict rules, i guess we should avoid confusing users. negative nuttx errno values badly conflict with MBEDTLS_ERR_xxx values and confuses mbedtls_strerror. for [...]

I get it. Let me map these to mbedtls error

closed 2025-07-30 by xiaoxiang781216