apache/nuttx-apps #3146: [FEATURE] nxboot.h location

TimJTi, 2025-07-29:
### Is your feature request related to a problem? Please describe.

The nxboot.h header file is currently located in `apps/boot/nxboot/include` directory

Should this not be moved to `apps/include/boot `(or a similarly named subdirectory of apps/include to make it easier for  apps - including custom "out-of-tree" apps - to include it without having to use convoluted paths in the #include statement?

The header has a number of public function definitions that - as I understand it - would typically be needed by user apps, like:

```
int nxboot_get_state(struct nxboot_state *state);
int nxboot_confirm(void);
int nxboot_get_confirm(void);
int nxboot_open_update_partition(void);
```

It does also include  some most likely "non-public" definitions such as:

```
int nxboot_ramcopy(void);
void nxboot_progress(enum progress_type_e type, ...);
```

and enums etc. So we probably need a "local" header file as well as a "public" one?

@michallenc - thoughts?

### Describe the solution you'd like

I am happy to work on the PR after input and suggestions from others, if this is agreed to be a good change.

### Verification

- [x] I have verified before submitting the report.

michallenc, 2025-07-29:
I am not against moving the header to `apps/include/boot`, there was no particular reason why I included it in `apps/boot/nxboot/include/`, just kept the same style used already for mcuboot. And it's true moving this to common include directory would make more sense as it would follow other examples.

Regarding `nxboot_progress`, I think it should be public API as well. The idea is you don't have to use `boot/nxboot/nxboot_main.c` as an entry point, but you can write your own if you want/need to.

Regarding convoluted paths, it shouldn't be necessary. `Make.defs` in `apps/boot/nxboot`

```
ifneq ($(CONFIG_BOOT_NXBOOT),)
CONFIGURED_APPS += $(APPDIR)/boot/nxboot

CFLAGS += ${INCDIR_PREFIX}$(APPDIR)/boot/nxboot/include

endif
```

should ensure the header is installed and only `<nxboot.h>` include is required. The move to `apps/include/boot` would prolong it to `<boot/nxboot.h>`, but it's not a big deal.

TimJTi, 2025-07-29:
Thanks @michallenc 

I am definitely not a makefile/make.defs expert in any way shape or form, but I have a "MyApps" directory under nuttx/apps (I used to have a symlink located there to an out-of-tree folder but that caused other issues) and I cannot seem to include header files unless they are there in the apps/include folders with very explicit paths (i.e. ../apps/nuttx/boot/nxboot/include/nxboot.h. PITA!!

If it's easily sorted by some other means that makes total sense and it probably needs to be in the makefiles/make.defs that live under MyApps but I am at a loss to find the magic incamtation lol. I'll have another crack tomorrow.

TimJTi, 2025-07-30:
No idea what I did yesterday: `#include <nxboot.h>` certainly works today and I can only think that was the only variant I didn't try or something equally dumb. <embarrassed sigh>

This isn't broken...we'll leave it as is :)